ArduPilot/MAVProxy #403: high memory usage

kd0aij, 2017-04-10:
mavexplorer running on Ubuntu 16.04 with 2 graphs displayed (over 600MB consumed per graph):
[image]

stephendade, 2017-04-11:
How large was the log file you had running in MAVExplorer. 

It does use a lot of memory with larger logfiles

kd0aij, 2017-04-11:
The log file was about 30MB, quite small compared to the memory usage.

peterbarker, 2017-04-13:
On Tue, 11 Apr 2017, Mark Whitehorn wrote:

> The log file was about 30MB, quite small compared to the memory usage.

There are two problems.  The first is this 20-to-1 size expansion.

The second is that every graph you create creates an out-and-out copy of 
the data.

Both need fixing :-)

swaroophangal, 2018-08-03:
Hello. My logfile size is ~250MB. I love this tool, but my machine conks off everytime i run this tools with a logfile of this size. Any other suggestions with the feature set as that of this tool? Or any other route to analyze data of that size?

peterbarker, 2018-08-05:
On Fri, 3 Aug 2018, swaroophangal wrote:

> Hello. My logfile size is ~250MB. I love this tool, but my machine conks off everytime i run this tools with a logfile of this size. Any other suggestions
> with the feature set as that of this tool? Or any other route to analyze data of that size?

I'm assuming you're talking MAVExplorer here :-)

We're *quite* aware of the memory problems - a lot of swearing has been 
done over the matter....

You have a few options.  One is to trim the log to pieces of interest - 
there's mavextract.py and mavlogdump.py's `--condition` option to do that.

Most people point to apmplanner2 as the best graphing tool out there - you 
might give that a shot.

There's also the "dron.ee" tool now-adays, but I haven't played with 
that...

amilcarlucas, 2018-10-17:
The latest version uses a lot less memory

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas