ArduPilot/MAVProxy #406: Reset state estimate (for SITL)

madratman, 2017-04-14:
I am using the ardupilot_gazebo_sitl_plugin to explore some learning algorithms in a gazebo environment. So, every time my drone is about to crash, I would like to "reset" it completely. I tried various approaches like going to mode STABILIZE, sending rc[3] = 0, and then resetting the gazebo model. Another way is to set the mode to RTL, wait to land and then reset model. Either way, the copter thinks it is in the previous state after resetting the gazebo model, and hence, I want to find a way to hardcode/reset the pose itself. Is there a way to do the same over mavproxy?

moritzschaefer, 2019-11-20:
I am also still looking for a simple way to just reset the copter to its initial position

Kinggrass, 2023-07-17:
Also looking for a solution.