ArduPilot/MAVProxy #413: Could you make a python archive of this?

marcmerlin, 2017-05-16:
I've just spend 2-3H trying to get it working after it stopped due to some other python upgrade in some math/graph lib.
So I upgraded everything I could.
Then I followed the instructions from http://ardupilot.org/dev/docs/using-mavexplorer-for-log-analysis.html
It says "after running that you should have a new command “MAVExplorer.py” in your path."
That's not true, pip doesn't install the actual MAVExplorer.py
So I cloned this repo, then it wouldn't run because
saruman:~/RC/MAVProxy$ ./MAVExplorer.py 
Traceback (most recent call last):
  File "./MAVExplorer.py", line 19, in <module>
    from MAVProxy.modules.lib.graph_ui import Graph_UI
Ok, I moved everything to /usr/local/lib/python2.7/dist-packages//MAVProxy
Now that part runs, but I'm stuck on
  File "./MAVExplorer.py", line 19, in <module>
    from MAVProxy.modules.lib.graph_ui import Graph_UI
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/graph_ui.py", line 1, in <module>
    from MAVProxy.modules.lib import grapher
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy/modules/lib/grapher.py", line 10, in <module>
    import matplotlib
  File "/usr/lib/python2.7/dist-packages/matplotlib/__init__.py", line 1184, in <module>
    rcParams = rc_params()
  File "/usr/lib/python2.7/dist-packages/matplotlib/__init__.py", line 1027, in rc_params
    return rc_params_from_file(fname, fail_on_error)
  File "/usr/lib/python2.7/dist-packages/matplotlib/__init__.py", line 1160, in rc_params_from_file
    if key not in _all_deprecated])
  File "/usr/lib/python2.7/dist-packages/matplotlib/__init__.py", line 909, in __init__
    self[k] = v
  File "/usr/lib/python2.7/dist-packages/matplotlib/__init__.py", line 928, in __setitem__
    cval = self.validate[key](val)
  File "/usr/lib/python2.7/dist-packages/matplotlib/rcsetup.py", line 853, in validate_cycler
    cycler_inst.change_key(prop, norm_prop)
AttributeError: 'Cycler' object has no attribute 'change_key'

This is a game of whack-a-mole, I'm back to problem with the math lib :(
Given that getting this working on a random system is a mess, would you consider just shipping a python archive that contains all your libraries 
Either this http://stackoverflow.com/questions/2026395/how-to-create-python-egg-file
or a runimage, or anything :)

marcmerlin, 2017-05-16:
Hi @tridge do you still maintain this? If so, can you think of a way to make a prepackaged distribution of it considering how many libraries it depends on and how they keep changing in incompatible ways, in turn breaking mavexplorer repeatedly?

peterbarker, 2017-05-16:
On Tue, 16 May 2017, Marc MERLIN wrote:

> I've just spend 2-3H trying to get it working after it stopped due to some
> other python upgrade in some math/graph lib.
> So I upgraded everything I could.
> Then I followed the instructions from
> http://ardupilot.org/dev/docs/using-mavexplorer-for-log-analysis.html

What operating system are you using here?

I've noticed it is possible to get into troubles when installing python 
packages using both pip and the sytem package manager.  May be worth 
checking if python-matplotlib is installed as a system package; if so, you 
could try "apt-get remove -y matplotlib; pip install -y matplotlib"

marcmerlin, 2017-05-17:
@peterbarker I'm running debian/testing.
saruman:~# pip install  matplotlib
Requirement already satisfied: matplotlib in /usr/lib/python2.7/dist-packages

I'm attaching this just in case this helps:
saruman:~# dpkggrep python | grep -v deinstall > /tmp/python.txt
[python.txt](https://github.com/ArduPilot/MAVProxy/files/1006515/python.txt)

I'm happy to try and fix my system if I even know what to fix, but from what I can tell, it's hard to get python programs that require a bunch of libs, to work on a variety of systems, so making a python package that includes all the lib versions it actually likes, would make things easier :)
In the meantime, I'm happy to try and see why it's not working on my system.

peterbarker, 2017-05-17:
On Tue, 16 May 2017, Marc MERLIN wrote:

> In the meantime, I'm happy to try and see why it's not working on my system.

Try upgrading the "cycler" python package?

Having lived through the zlib-security-issues time period, I'm no great 
fan of incorporating dependencies into the end-user tools, BTW :-)

stephendade, 2017-05-17:
Comparing the documentation at http://ardupilot.org/dev/docs/using-mavexplorer-for-log-analysis.html and http://ardupilot.github.io/MAVProxy/html/getting_started/download_and_installation.html, we should edit the Ardupilot documentation to just include a link to the MAVProxy site.

Saves potential confusion (and makes maintenance easier)

marcmerlin, 2017-05-17:
Thanks @stephendade I indeed ended up with the wrong documentation, +1 on removing the other one and linking to the up to date one.

marcmerlin, 2017-05-17:
@peterbarker  thank you, 
apt-get install python-cycler
fixes the problem indeed. Now MAVExplorer.py launches with
MAVExplorer.py 7.BIN 
07:08:49: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and wxPython used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
MAV> 
Hopefully the warning is indeed just a warning :)
And on lib dependencies and static vs dynamic linking (which this kind of amounts to), I hear you, yes I've seen lib security updates too, but let's be honest, the risk and impact are probably minor in the case of MAVExplorer
As a counter example, I'll point you to https://subsurface-divelog.org/ which used to be sometimes hours of hair pulling to build on my system. The amount of libraries, sub libraries and weird dependencies fighting with my own system dependencies and build libs that you could only have one version of, were just a disaster almost every time I tried to build it. Nowi it's built and distributed with appimage, and it just works.
Back to python, let's say I have just another single program that uses those libs. By updating them all to work with MAVExplorer, I'd now have broken that other program.
Python libs that are incompatible between versions and aren't versioned so that you can have one of each fo the incompatible versions for each program you have and uses them, is bad news IMO.
All that said, while I'm not a python expert, I thought that's what python egg was for.

Either way, thanks for your help, I'm back in business for now :)

amilcarlucas, 2018-10-17:
Windows builds are now available, so this can be closed now

amilcarlucas, 2018-10-17:
you can do a `python setup.py sdist` that will generate you a .whl python file that you can install on other machines using `pip install Mavproxy-1-7-x.whl`
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas