ArduPilot/MAVProxy #417: MAVExplorer not opening AC3.5-rc7 logs

auturgy, 2017-05-29:
MAVExplorer has "Issue loading file" with *.bin and *.log from rc7. Logs are not corrupted, and open in MP-latest for log analysis.

peterbarker, 2017-05-29:
Supply a sample, please.

auturgy, 2017-05-29:
https://drive.google.com/drive/folders/0B21eBYNmOfAKQjV2ZU94S2t1dnc
None of these open in mavexplorer (win10)

peterbarker, 2017-05-29:
No problems under Linux.  Please post the text of the error.

auturgy, 2017-05-29:
"Error loading file" in console

peterbarker, 2017-05-29:
That's not... particularly informative :-)  Is there a stack trace on stdout?

peterbarker, 2017-05-29:
Actually, that's only called if no argument is supplied.  There was a problem when logs had spaces in the filename - is that the case for the files you're supplying?

auturgy, 2017-05-29:
they're the generated filenames - I'll check

auturgy, 2017-05-29:
yes, spaces in the filename between date and time.

auturgy, 2017-05-29:
I just renamed one - still no open, same brief error msg.

stephendade, 2017-06-02:
I've renamed the files to have no spaces in them. They open/graph for me. Using MAVProxy 1.4.42 on Win10

auturgy, 2017-06-02:
I think windows hates me. Might just build up a Linux box and save the frustration.

Regards,

James

peterbarker, 2017-06-05:
@auturgy 

https://github.com/peterbarker/MAVProxy/commit/2f21cfc1ae8c08ecbb30bdc5157a6a5a1c0ee1d7#diff-e6b1d8e5a4cc4e1a493c3da49b991ab5R431

That patch should have made the error message longer.  It did, didn't it?

auturgy, 2017-06-05:
> @auturgy
>
> peterbarker/MAVProxy@2f21cfc#diff-e6b1d8e5a4cc4e1a493c3da49b991ab5R431
>
> That patch should have made the error message longer. It did, didn't it?

Yes: posted the output in gitter on Saturday 

Regards,

James

auturgy, 2017-11-15:
Closing.  Unsure what changed, but now working.  Suggest operator error or windows being windows.

closed 2017-11-15 by auturgy