ArduPilot/MAVProxy #434: No emergency stop on wireless disconnect

J-tt, 2017-08-03:
I'm using a wireless rover set up that roves between access points, and have had issues that when the wireless drops out, the rover continues forward.

I had a bit of a crash, but everything is alright.

It would seem logical to have an automatic stop when wireless drops out.

gmorph, 2017-08-03:
This isn't a bug but rather a feature request.
Further, the code supports going into failsafe when RC or Telemetry connection is lost.  Please see the ArduPIlot website for further details.

Thanks.

closed 2017-08-03 by peterbarker

J-tt, 2017-08-04:
@gmorph I've been searching through the mavproxy documentation, but I can't see any options for a failsafe when a controlling computer (eg: a PC with Mission Planner) loses contact with the device running mavproxy. 

If you could point me in the right direction that would be awesome!

gmorph, 2017-08-05:
Hmmmm, the Rover Doco appears to have lost the Failsafe (FS) Page.  That is a bug in the doco.  I'll fix it later.
The parameters page has all the info but not an explanation of how to do it as such.
http://ardupilot.org/rover/docs/parameters.html
You want to set FS_ACTION to 2 so it goes into HOLD mode when the FS triggers.
Set FS_TIMEOUT for how long you want to wait after loosing connection before FS triggers.  Default is 5 seconds.
And finally set FS_GCS_ENABLE to 1.  When the Rover looses connection with the GCS this will mean it will go into FS after FS_TIMEOUT number of seconds and perform FS_ACTION.

Better to ask this kinds of questions on the discuss server
discuss.ardupilot.org

Thanks, Grant.

J-tt, 2017-08-06:
@gmorph Thank you for your help!