ArduPilot/MAVProxy #438: MAVProxy - no reference to origins of MAVProxy

auturgy, 2017-08-10:
URL: ./index
Title: MAVProxy

just a note to spark discussion about putting a PR in: I think it would be good to mention something along the lines of "MAVProxy was first developed by CanberraUAV, to enable the use of companion computing and multiple datalinks with ArduPilot.  It has grown to be one of the most versatile tools in the ArduPilot ecosystem, and many of the features users now see in MissionPlanner can trace their origins to MAVProxy.
The team at CanberraUAV still lead development of MAVProxy: they are a volunteer organisation, but would appreciate your support <Patreon Link>

stephendade, 2017-08-10:
Would be useful to put it in the readme.md too

stephendade, 2017-09-16:
Fixed in documentation at http://ardupilot.github.io/MAVProxy/html/index.html

closed 2017-09-16 by stephendade