ArduPilot/MAVProxy #451: Download and Installation -

ShingoMatsuura, 2017-09-28:
URL: ../getting_started/download_and_installation
Title: Download and Installation linsux

Please add this pip commad before pip insall Mavproxy

pip install future lxml

stephendade, 2017-10-02:
The future and lxml packages are usually installed as part of pymavlink, which in turn is installed as part of MAVProxy. Could you post the exact steps that caused your error?

ShingoMatsuura, 2017-10-02:
Thank you for your comment. I followed this site steps.

http://ardupilot.github.io/MAVProxy/html/getting_started/download_and_installation.html#linux

stephendade, 2017-10-02:
Which operating system and version? Are you able to post the output of the commands? - it would help in diagnosing the issue.

auturgy, 2017-11-15:
Suggest close.  Updates have been released to both pymavlink and mavproxy.

closed 2017-11-15 by peterbarker