ArduPilot/MAVProxy #456: Possible bug with wx 3 gtk 2

jaredd, 2017-10-13:
I ran into a problem running the ArduCopter sitl with the map display using the following set of software packages.  I was getting a TypeError from a call in MAVProxy to some wxPython module.  I was able to find the call and determine a workaround fix that gets everything running without error but with unknown repercussions to ardupilot as a whole.

Some info:
```
$ uname -a
Linux linux-xdol 4.13.4-1-default #1 SMP PREEMPT Wed Sep 27 14:20:45 UTC 2017 (4dec972) x86_64 x86_64 x86_64 GNU/Linux

$ wx-config --version-full --selected-config
3.0.1.0
gtk2-unicode-3.0

$ git describe
ArduCopter-3.1.0-rc6-19277-g6bd5ac78f

 in ardupilot/ArduCopter directory
$ sim_vehicle.py --console --map --aircraft test
...
BUILD SUMMARY
Build directory: /home/jaredd/pyvenvs/gnuradio3.7/src/ardupilot/build/sitl
Target          Text     Data  BSS    Total  
---------------------------------------------
bin/arducopter  1534793  2130  37408  1574331

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (2.832s)
...
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
no script test/mavinit.scr
Log Directory: test/logs/2017-10-11/flight13
Telemetry log: test/logs/2017-10-11/flight13/flight.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
Process Process-2:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib64/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/home/jaredd/pyvenvs/gnuradio3.7/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 92, in child_task
    self.app.frame = MPSlipMapFrame(state=self)
  File "/home/jaredd/pyvenvs/gnuradio3.7/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap_ui.py", line 47, in __init__
    self.SetMenuBar(self.menu.wx_menu())
  File "/home/jaredd/pyvenvs/gnuradio3.7/lib/python2.7/site-packages/MAVProxy/modules/lib/mp_menu.py", line 246, in wx_menu
    menubar.Append(m.wx_menu(), m.name)
  File "/home/jaredd/pyvenvs/gnuradio3.7/lib/python2.7/site-packages/MAVProxy/modules/lib/mp_menu.py", line 190, in wx_menu
    m._append(menu)
  File "/home/jaredd/pyvenvs/gnuradio3.7/lib/python2.7/site-packages/MAVProxy/modules/lib/mp_menu.py", line 139, in _append
    menu.Append(-1, self.name, submenu)
  File "/home/jaredd/pyvenvs/gnuradio3.7/lib/python2.7/site-packages/wx-3.0-gtk2/wx/_core.py", line 12003, in Append
    return _core_.Menu_Append(*args, **kwargs)
TypeError: String or Unicode type required
STABILIZE> Unloaded module map
...
```
I ended up making the following changes, the diff is with the latest MAVProxy/modules/lib/mp_menu.py from the master branch on the web site so I marked my own mods with ```### <<< MODIFIED```.

```
$ diff mp_menu.py ~/Downloads/mp_menu.py 
9a10
> import platform
139c140
<         menu.Append(-1, self.name, "", 0)   ### <<< MODIFIED
---
>         menu.AppendMenu(-1, self.name, submenu)
204,205c205,208
< 
<         menu.Append(-1, self.name, "", 0)   ### <<< MODIFIED
---
>         if platform.system() == 'Darwin':
>             menu.Append(-1, self.name, self.wx_menu()) #use wxPython_phoenix
>         else:
>             menu.AppendMenu(-1, self.name, self.wx_menu())
284c287
<         return dlg.GetPath()
---
>         return dlg.GetPath().encode('utf8')
```

peterbarker, 2017-10-17:
On Thu, 12 Oct 2017, JaredD wrote:

> I ran into a problem running the ArduCopter sitl with the map display using
> the following set of software packages. I was getting a TypeError from a
> call in MAVProxy to some wxPython module. I was [...]

Thanks for the report!

It would be awesome of you could create a pull request with these changes 
in it - I have concerns that trying to transcribe your (well-descibed) 
changed below) may result in a subtle change which doesn't help fix the 
problem!

If you can come over to our gitter channel then I can help you do it: 
https://gitter.im/ArduPilot/MAVProxy

Yours,
-- 
Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams

jaredd, 2017-10-18:
Not sure if this is useful, but I updated my git repo and tried from the repo directory

$ pip install -U .

This installed from the repo and I was able to start up my arducopter sim without error.  So perhaps my problems were fixed via an update between the downloaded version and the repo version.

closed 2017-10-18 by jaredd