ArduPilot/MAVProxy #467: Pymavlink conflicts

stephendade, 2017-12-02:
Based upon this comment:
https://github.com/ArduPilot/MAVProxy/commit/b704fb901fc87a21b04909ba2811cf86f4d0baa0

After further research, there appears to be some confusion about which mavlink we're supposed to use with MAVProxy.

From Ardupilot's Mavlink (https://github.com/ArduPilot/mavlink/commit/f4ef4fb7bdd43b5f110d81953057126b15c1a58c), `MAV_TYPE_DODECAROTOR` has a value of `28`

However, Lorenz's Mavlink (https://github.com/mavlink/mavlink/commit/bd9b88d6a6014adb11a725ac24e190e152a9f8ec) has  `MAV_TYPE_PARAFOIL` has a value of `28`

Looking at pymavlink on Pypi (https://pypi.python.org/pypi/pymavlink), Version 2.2.6 has  `MAV_TYPE_PARAFOIL` has a value of `28`. So Lorenz's pymavlink is in the Pypi repo. Or there's been some really strange git merging going on between the Ardupilot's and Lorenz's pymavlink.

Having essentially two different pymavlinks is going to make it super confusing and difficult for any GCS's (ie MAVProxy)  out there.

In MAVProxy's case, it will crash if started up using copter SITL and using the Pypi pymavlink.

Should MAVProxy be able to support different implementations of pymavlink? Or we just tell the users to only use the Ardupilot pymavlink?

tridge, 2017-12-02:
thanks! I've done a new pymavlink release with this fixed

amilcarlucas, 2018-10-17:
@tridge please close this one

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas