ArduPilot/MAVProxy #468: Some version to python3.6?

francisrod01, 2017-12-05:
Some version to python3.6?

SamuelDudley, 2017-12-05:
Im not entirely sure what the issue is but MAVProxy currently does not run under python 3 if that is what you are attempting to do. See https://github.com/ArduPilot/MAVProxy/pull/420

francisrod01, 2017-12-05:
Thanks, @SamuelDudley!
That is the point!

I'll follow the issue about it.

francisrod01, 2017-12-07:
@SamuelDudley after the rebase, I try to upgrade mavproxy as `pip3` in my hasperryPI VirtualMachine in VMBox and it's fail.

```
copying ./dialects/v20/slugs.xml -> build/lib.linux-i686-3.4/pymavlink/dialects/v20
    running build_ext
    building 'mavnative' extension
    creating build/temp.linux-i686-3.4
    creating build/temp.linux-i686-3.4/mavnative
    i586-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -Igenerator/C/include_v1.0 -Igenerator/C/include_v2.0 -Imavnative -I/usr/include/python3.4m -c mavnative/mavnative.c -o build/temp.linux-i686-3.4/mavnative/mavnative.o
    mavnative/mavnative.c:8:20: fatal error: Python.h: No such file or directory
     #include <Python.h>
                        ^
    compilation terminated.
    error: command 'i586-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-z166c4hi/pymavlink/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-kawqmgqx-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-z166c4hi/pymavlink/
```

SamuelDudley, 2017-12-07:
Can't say I'm using pip to test my branches sorry. The stack trace looks like it's failing during the pymavlink mavnative build process. Not being able to find python.h usually means you haven't installed the python-dev packages or they can't be found...

SamuelDudley, 2017-12-07:
https://ubuntuforums.org/showthread.php?t=2074355 
Run `sudo apt-get install python3-dev` and then try again

francisrod01, 2017-12-08:
Thanks, it works!
But now using `pip3` to install `mavproxy` I see that error `

Successfully installed mavproxy-1.6.2

```
$ mavproxy.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14551 --out udp:10.55.222.120:14556
Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 12, in <module>
    import serial, Queue, select
ModuleNotFoundError: No module named 'Queue'
```
This package in pip3 is it outdated?

Sorry about some questions, but I did not want to have to clone repository and install global on system.

Thank you for understand! :)

SamuelDudley, 2017-12-09:
Hi,
The problem is that MAVProxy doesn't run under python3 and you are trying to install the unmodified version from pip. You need to use my git branch which has the required changes to run with python3. 
I haven't used pip this way but a quick google turns up this result https://stackoverflow.com/questions/20101834/pip-install-from-git-repo-branch

francisrod01, 2017-12-09:
Thank you @SamuelDudley , installing remotely it new to me!

It running with `pip3 install ~/MAVProxy` local or `git+https://github.com/SamuelDudley/MAVProxy@wx_phoenix_wip_rebase`.

Well, in my first test, it was necessary to install `numpy` dependency of MAVProxy.

I haven't found `terrain` and `adsb`  compatibles dependencies to use with MAVProxy package.

I received the output before my first test with dronekit copter example:

```
$ mavproxy.py --master tcp:127.0.0.1:5760 --out udp:127.0.0.1:14551 --out udp:10.55.222.120:14556
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect tcp:127.0.0.1:5760 source_system=255
Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 407, in import_package
    mod = importlib.__import__(name)
  File "<frozen importlib._bootstrap>", line 1059, in __import__
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_adsb.py", line 11, in <module>
    from MAVProxy.modules.mavproxy_map import mp_slipmap
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 13, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 280, in load_module
    m = import_package(modpath)
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 410, in import_package
    mod = importlib.__import__(name)
  File "<frozen importlib._bootstrap>", line 1059, in __import__
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 950, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_adsb.py", line 11, in <module>
    from MAVProxy.modules.mavproxy_map import mp_slipmap
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_map/mp_slipmap.py", line 13, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 407, in import_package
    mod = importlib.__import__(name)
  File "<frozen importlib._bootstrap>", line 1059, in __import__
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 948, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'adsb'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 280, in load_module
    m = import_package(modpath)
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 410, in import_package
    mod = importlib.__import__(name)
  File "<frozen importlib._bootstrap>", line 1059, in __import__
  File "<frozen importlib._bootstrap>", line 978, in _gcd_import
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load
  File "<frozen importlib._bootstrap>", line 948, in _find_and_load_unlocked
[... 7 lines cut ...]
STABILIZE> Mode STABILIZE
Exception in thread main_loop:
Traceback (most recent call last):
  File "/home/data-science/anaconda3/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/data-science/anaconda3/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 757, in main_loop
    master.wait_heartbeat()
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/mavutil.py", line 383, in wait_heartbeat
    return self.recv_match(type='HEARTBEAT', blocking=blocking)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/mavutil.py", line 344, in recv_match
    m = self.recv_msg()
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/mavutil.py", line 318, in recv_msg
    msg = self.mav.parse_char(s)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9257, in parse_char
    self.__callbacks(m)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9234, in __callbacks
    self.callback(msg, *self.callback_args, **self.callback_kwargs)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_link.py", line 379, in master_callback
    mavutil.mavlink.MAV_TYPE_DODECAROTOR]:
AttributeError: module 'pymavlink.dialects.v10.ardupilotmega' has no attribute 'MAV_TYPE_DODECAROTOR'
```

So.. some errors but it works!! :1st_place_medal:

SamuelDudley, 2017-12-09:
Working backwards the errors you are seeing are: a pymavlink issue https://github.com/ArduPilot/MAVProxy/issues/467 try updating pymavlink to the very latest release.
The other two are both due to opencv python bindings that you don't have installed.

SamuelDudley, 2017-12-09:
I'm building my python bindings from source, so if you wanted to go down that route this may help: https://www.pyimagesearch.com/2017/09/04/raspbian-stretch-install-opencv-3-python-on-your-raspberry-pi/

otherwise you could try the unnoficial python bindings in pip `pip3 install opencv-python` : https://pypi.python.org/pypi/opencv-python

francisrod01, 2017-12-09:
Thanks! Some errors are solved but I receive others.

I installed `future`, `xml`, `opencv-python` and upgrade `pymavlink` dependencies but I still receive some errors like this:

```
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from tcp:127.0.0.1:5760
online system 1
STABILIZE> Mode STABILIZE
Exception in thread main_loop:
Traceback (most recent call last):
  File "/home/data-science/anaconda3/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/home/data-science/anaconda3/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 757, in main_loop
    master.wait_heartbeat()
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/mavutil.py", line 383, in wait_heartbeat
    return self.recv_match(type='HEARTBEAT', blocking=blocking)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/mavutil.py", line 344, in recv_match
    m = self.recv_msg()
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/mavutil.py", line 318, in recv_msg
    msg = self.mav.parse_char(s)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9257, in parse_char
    self.__callbacks(m)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 9234, in __callbacks
    self.callback(msg, *self.callback_args, **self.callback_kwargs)
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_link.py", line 379, in master_callback
    mavutil.mavlink.MAV_TYPE_DODECAROTOR]:
```

Should I uninstall the modemmanager from the system?

This time I'm testing with ubuntu.
I was unable to install python3.6 on Raspibian because of lack of system space.

stephendade, 2017-12-09:
>Should I uninstall the modemmanager from the system?

Not mandatory, but it does sometimes conflict with MAVProxy when using serial ports.

The other error you're getting is due to using an old version of pymavlink - you'll need to upgrade to the latest version (2.2.7). Use `pip install --upgrade pymavlink`

francisrod01, 2017-12-09:
I'm using a virtualenv, so.. `--user` not available here.

```
(ardupilot) data-science@datascience-VirtualBox:~$ pip3 show pymavlink
Name: pymavlink
Version: 2.2.7
Summary: Python MAVLink code
Home-page: https://github.com/ArduPilot/pymavlink/
Author: UNKNOWN
Author-email: UNKNOWN
License: LGPLv3
Location: /home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages
Requires: lxml, future
```

I updated pymavlink to 2.2.7 and I receive a loop of the same errors:

```
STABILIZE> Mode STABILIZE
fence breach
Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 737, in periodic_tasks
    m.idle_task()
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_terrain.py", line 141, in idle_task
    self.send_terrain_data()
FileNotFoundError: [Errno 2] No such file or directory: '/home/data-science/.tilecache/SRTM/filelist_python'
Traceback (most recent call last):
  File "/home/data-science/envs/python3.6/ardupilot/bin/mavproxy.py", line 737, in periodic_tasks
    m.idle_task()
  File "/home/data-science/envs/python3.6/ardupilot/lib/python3.6/site-packages/MAVProxy/modules/mavproxy_terrain.py", line 141, in idle_task
    self.send_terrain_data()
FileNotFoundError: [Errno 2] No s
```

francisrod01, 2017-12-26:
I think that it's not prepare to run in python3.6.
Some dependencies aren't resolve yet.

francisrod01, 2018-02-26:
I'm expecting to use this version of the ArduPilot using MAVProxy with Python 3.6.x but until now this is not official, right?

Zhenbin24, 2018-04-10:
can't running python3.6 ?  how to handle it?
[image]

Zhenbin24, 2018-04-10:
@francisrod01 @SamuelDudley @stephendade @benizl @lucasdemarchi

francisrod01, 2018-04-11:
Issues related: 
https://github.com/ArduPilot/MAVProxy/issues/280
https://github.com/ArduPilot/MAVProxy/pull/420
https://github.com/ArduPilot/MAVProxy/pull/420

amilcarlucas, 2018-10-17:
Try the latest version, it should now support python 3

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas