ArduPilot/MAVProxy #469: Cannot use OFFBOARD flight mode.

weiminshen99, 2017-12-06:
I am using MAVProxy to control a PX4, and would like to switch the flight mode to OFFBOARD. However, when I do so, MAVProxy reports that this mode is UNKNOWN. How do I fix that? Thanks!

SamuelDudley, 2017-12-07:
Am I correct in thinking that this is just a case of MAVProxy not being aware of the mode number in the heartbeat message? To confirm, although MAVProxy displays `UNKNOWN` is the autopilot actually in `OFFBOARD` mode?

weiminshen99, 2017-12-07:
Thanks for the relay. Yes, that is correct. Because MAVProxy cannot recognize the mode and display it as UNKNOWN, my control program cannot detect it properly. Also, the HAEATBEAT message was also confused how to deal with that mode. 

By the way, I tried the solution/fix from "Pass mode through to set_mode #302," but I am not sure how to make the change in the system so that the mode can be recognized properly.

peterbarker, 2017-12-07:
On Thu, 7 Dec 2017, weiminshen99 wrote:

> Thanks for the relay. Yes, that is correct. Because MAVProxy cannot
> recognize the mode and display it as UNKNOWN, my control program cannot
> detect it properly. Also, the HAEATBEAT message was also confused how to
> deal with that mode.

Are you running the code from the github repositories or from pip?

weiminshen99, 2017-12-08:
I have tried both pip and github repository, with the fix of "Pass mode through to set_mode #302" in place. 
Here is the output of mavproxy:
MANUAL> mode OFFBOARD
MANUAL> ('-------------Mode Requested: ', 'OFFBOARD') 
Got MAVLink msg: COMMAND_ACK {command : 176, result : 0}
UNKNOWN> Mode UNKNOWN

Here is the output observed from my control program:
>>> Exception in message handler for HEARTBEAT
>>> mode 393216 not available on mavlink definition

weiminshen99, 2017-12-21:
After much search and testing, here is the conclusion for people who are interested:
1. pymavlink 2.0.6 will not work, and the problem is as I experienced and reported;
2. pymavlink 2.2.4 will work with the OFFBOARD mode. However, this version still does not support some of the newer mode such as TAKEOFF, but that does not matter much for me.
In summary, MAVProxy with pymavlink2.2.4 will support offboard, and I tested on my UAV with real flights.

Finally, I would like to thank you all for the feedback and helps you gave me over the process.  I hope this shared experience will be useful for people in the future. Thanks!

closed 2017-12-21 by weiminshen99