ArduPilot/MAVProxy #474: DGPS Inject & forwarding packet loss

antonvh, 2017-12-20:
Hi,

I have the following setup:
- An Intel Edison with Yocto running Mavproxy.
- It has an RFD900 (868 in fact) modem on the serial port
- It runs a wifi hotspot
- It runs a DGPS base station, serving out RTCM3 via TCP on port 9000.

I can just run mavproxy with `--out=udpbcast:192.168.42.255:14550` and every device that runs QGCS on the hotspot will be able to connect to the drone. Yay!

However, as soon as I start to inject DGPS packages the connected device get a connection loss every second second. Either with `>>module load DGPS_TCP` or with `--load-module DGPS_TCP`

The official DGPS code in this repo was UDP an thus unusable, so I had to adapt it to TCP for my dgps server. My code is here:
https://github.com/antonvh/MAVProxy/blob/reach_inject/MAVProxy/modules/mavproxy_DGPS_TCP.py

How can I start to find out where the intermittent connection loss is coming from?

antonvh, 2018-01-08:
The problem seems to be with the gps_rtcm_data_send() command. It breaks every other packet relay via the proxy. I don't know where to debug it.

My packet sizes are really small: 27 bytes. Flags: 0b1011000

matislovas, 2018-04-23:
Hi,

 Have you figured it out? I'm trying to adapt your code, since RTKLIB's cmd line tool str2str can only transmit TCP packets, not UDP.

 When I load your module, I'm getting some of these lines:

Sending DGPS RTCM3 data flags 0b1001011, length 192
Sending DGPS RTCM3 data flags 0b1010001, length 615
Sending DGPS RTCM3 data flags 0b1010011, length 615
 Is it ok?

antonvh, 2018-05-02:
That's normal debug behavior. It would be neater if it would take the verbosity level of mavproxy into account. I fixed the timeout issue too in the mean time.

closed 2018-05-02 by antonvh

alduxvm, 2018-09-01:
This might be out of scope, but you might be able to help me, how do your 'DGPS base station'?? are you using RTKLIB? or what software do you use in the base station to read a ublox via serial port (usb)

antonvh, 2018-09-11:
I use ReachRTK. (Reachview)