ArduPilot/MAVProxy #476: MAVProxy ignores --mav20

alexx600, 2017-12-21:
Hello,
When I run SITL using
`sudo python sim_vehicle.py -v ArduCopter -m "--mav20" -f hexa --console --map -L WIPB`
it should use MAVlink 2.0 Protocol 
As i Can assume "-m" means additional mavproxy arguments 

Despite of using --mav20 when i run some C code to read data i can see

```
Bytes Received: 34
Datagram: fe 1a a5 01 01 1b e4 81 6e 6e 00 00 00 00 00 00 00 00 17 fc 02 00 02 00 01 00 b7 00 37 00 c2 01 e9 b1 
Received packet: SYS: 1, COMP: 1, LEN: 26, MSG ID: 27
```

It looks like MAVlink 1.0== it starts with 0xfe

When i read data from QGroundControl it looks like
```
Bytes Received: 21
Datagram: fd 09 00 00 16 ff 00 00 00 00 00 00 00 00 06 08 c0 04 03 87 fa 
Received packet: SYS: 255, COMP: 0, LEN: 9, MSG ID: 0
```
Byte array starts with 0xfd so it is MAVlink2.0

Why  SITL ignore  "--ma20" commands ? What should change to use MAVlink 2.0 version?

peterbarker, 2017-12-21:
On Thu, 21 Dec 2017, alexx600 wrote:

> Hello,
> When I run SITL using
> sudo python sim_vehicle.py -v ArduCopter -m "--mav20" -f hexa --console
> --map -L WIPB
> it should use MAVlink 2.0 Protocol
> As i Can assume "-m" means additional mavproxy arguments

Have you changed the vehicle serial parameter to also specify mavlink2?

alexx600, 2017-12-22:
[image]
SERIAL0_PROTOCOL, SERIAL1_PROTOCOL and SERIAL2_PROTOCOL are MAVLINK2.0 
Maybe I should change another parameter ?