ArduPilot/MAVProxy #477: Non interactive option?

estechnical, 2017-12-21:
If I run mavproxy in docker, it is desirable to have a noninteractive foreground process. 

At the moment, if I run "mavproxy" as my process in the container, it exits with error 1 after a few seconds.

Running the docker container interactively, it continues to run ok. Is there an option eg "--non-interactive" or could one easily be made to suit small setups running in docker?

peterbarker, 2017-12-21:
On Thu, 21 Dec 2017, Ed Simmons wrote:

> If I run mavproxy in docker, it is desirable to have a noninteractive
> foreground process.
>
> At the moment, if I run "mavproxy" as my process in the container, it exits
> with error 1 after a few [...]

Well, there's `--daemon`....

estechnical, 2017-12-22:
Thanks, but, these should really be accessible as separate flags/options. The problem with the interactive cli is it will exit when stdin is detached by docker daemon.

Docker expects to run one foreground process in the container and ideally nothing gets detached from the main shell process. The `--daemon` option is unsuitable for docker because it backgrounds the process, but half of it's behaviour is desirable.

I'm not great at python, would you mind examining a PR to add this feature?
:)

peterbarker, 2017-12-22:
On Fri, 22 Dec 2017, Ed Simmons wrote:

> Thanks, but, these should really be accessible as separate flags/options.
> The problem with the interactive cli is it will exit when stdin is detached
> by docker daemon.
>
> Docker expects to run one [...]

No worries, PR away.  I'm a bit conused as to what you're expecting, 
'though - MAVProxy to stay attached but not read from stdin?!

I guess the PR will tell me :-)

estechnical, 2017-12-22:
Yes, basically a foreground version of the `--daemon` flag. If you look at eg apache or nginx web servers, these typically launch then detach from the controlling shell by default. Both provide some means to remain attached as a foreground process and are non interactive.

estechnical, 2017-12-22:
Thanks... Hopefully my simple changes in the PR do what I intended and nothing more ;)

closed 2018-03-06 by estechnical