ArduPilot/MAVProxy #481: command line output is erased

magicrub, 2018-01-11:
command line output is erased when pressing enter.

Example:
mavhelp about <press enter>
<output shows correctly>
<press enter>
<output disappears>

MAVProxy Version v1.6.2
OS: Windows7
Python 2.7.9
WXPython []

stephendade, 2018-01-12:
That's a known issue on Windows. See https://github.com/ArduPilot/MAVProxy/pull/416 for the fix.

Unfortunately, there's a few issues with the fix that mean it can't be merged in.

Overall issue is that Linux and Windows treat multi-threaded prints to the console very differently.

stephendade, 2020-09-06:
This has been fixed as of https://github.com/ArduPilot/MAVProxy/commit/90362c78fe2aa09687327ebf1d5b0e70974d5013

closed 2020-09-06 by stephendade