ArduPilot/MAVProxy #482: MAVCesium module not working

magicrub, 2018-01-11:
MAVCesium module not working.

 MAVProxy 1.6.2, 
Windows 7
Python 2.7.9

[image]

magicrub, 2018-01-11:
@peterbarker confirms he has cesium working on his Ubuntu system (unknown update method)
@kd0aij confirms his pip update to mavproxy on his Ubuntu system DOES NOT include cesium

SamuelDudley, 2018-01-11:
Hi, is that the traceback with moddebug == 3? If not can you please post the traceback of `module load cesium` after first entering `set moddebug 3`. Thanks!

magicrub, 2018-01-11:
that's with 3

magicrub, 2018-01-11:
I just tried again and now it says..
[image]

SamuelDudley, 2018-01-11:
Sorry, missed your last comment there... Can you check your MAVProxy pip install path and see if there are any files in the mavproxy_cesium directory (within the module directory).

magicrub, 2018-01-11:
for windows, I don't have one. For Cygwin pip is c\cygwin\bin\pip

magicrub, 2018-01-11:
[image]

magicrub, 2018-01-11:
seems a pip update actually did something.

kd0aij, 2018-01-11:
```
03:29:07 /usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_cesium$ ls
app  __init__.py  __init__.pyc
03:29:08 /usr/local/lib/python2.7/dist-packages/MAVProxy/modules/mavproxy_cesium$ ls app
api_keys.txt   cesium_web_server.py   __init__.py   static
api_keys.txt~  cesium_web_server.pyc  __init__.pyc  templates
```

magicrub, 2018-01-11:
my \programFiles(x86)\MAVProxy does not have a modules folder

SamuelDudley, 2018-01-11:
Interesting... I don't currently have access to a windows machine to debug for you but I have a Dell laptop on order. When that arrives I can try to reproduce your issue, however it seems to be related to the pip build and only on windows. How immediate is your fix requirement?

kd0aij, 2018-01-12:
I am on Ubuntu 16.04 and it looks like I have the same problem:
```
STABILIZE> module load cesium
STABILIZE> Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy-1.6.2-py2.7.egg/EGG-INFO/scripts/mavproxy.py", line 279, in load_module
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy-1.6.2-py2.7.egg/EGG-INFO/scripts/mavproxy.py", line 409, in import_package
ImportError: No module named mavproxy_cesium

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy-1.6.2-py2.7.egg/EGG-INFO/scripts/mavproxy.py", line 279, in load_module
  File "/usr/local/lib/python2.7/dist-packages/MAVProxy-1.6.2-py2.7.egg/EGG-INFO/scripts/mavproxy.py", line 409, in import_package
ImportError: No module named cesium

Failed to load module: No module named cesium. Use 'set moddebug 3' in the MAVProxy console to enable traceback
```

kd0aij, 2018-01-12:
MAVProxy-1.6.2-py2.7.egg does not contain the cesium module

stephendade, 2018-01-12:
@SamuelDudley  and @magicrub - I've not included cesium module in the Windows installer.

I've never actually tested the cesium module in Windows ... likely missing some pip libraries.

magicrub, 2018-01-12:
well, what pip libraries are needed? I'll happily install them manually to test!

SamuelDudley, 2018-01-12:
Only two additional libraries `tornado` and `configparser`

https://github.com/goodrobots/MAVCesium/blob/master/requirements.txt

Let me know how you go :)

SamuelDudley, 2018-01-12:
FYI you can also run MAVCesium as a seperate standalone application from MAVProxy if required / desired. Just execute https://github.com/goodrobots/MAVCesium/blob/master/app/cesium_web_server.py and provide a TCP/UDP/Serial MAVLink stream

mohammada315, 2021-06-30:
sorry
Traceback (most recent call last):
File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
self.run()
File "/usr/lib/python3.8/threading.py", line 870, in run
self._target(self._args, *self._kwargs)
File "/home/mohammad/.local/lib/python3.8/site-packages/MAVProxy-1.8.39-py3.8.egg/MAVProxy/modules/mavproxy_cesium/app/cesium_web_server.py", line 141, in main
File "/home/mohammad/.local/lib/python3.8/site-packages/MAVProxy-1.8.39-py3.8.egg/MAVProxy/modules/mavproxy_cesium/app/cesium_web_server.py", line 104, in start_app
File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/tcpserver.py", line 152, in listen
self.add_sockets(sockets)
File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/tcpserver.py", line 165, in add_sockets
self._handlers[sock.fileno()] = add_accept_handler(
File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/netutil.py", line 246, in add_accept_handler
io_loop = IOLoop.current()
File "/home/mohammad/.local/lib/python3.8/site-packages/tornado-6.2.dev1-py3.8-linux-x86_64.egg/tornado/ioloop.py", line 264, in current
loop = asyncio.get_event_loop()
File "/usr/lib/python3.8/asyncio/events.py", line 639, in get_event_loop
raise RuntimeError('There is no current event loop in thread %r.'
RuntimeError: There is no current event loop in thread 'Thread-1'.

Ryanf55, 2024-11-18:
It's working now. At least 3 devs have ran it successfully. This can be closed.

peterbarker, 2024-11-18:
Thanks @Ryanf55 !

closed 2024-11-18 by peterbarker