ArduPilot/MAVProxy #485: Reboot when connecting to board

mustaphos, 2018-01-17:
Hi.
Can i disable the reboot when i try to connect to the APM board?
I give power to the apm and it starts to work. 
But when i connect to it over serial using mavproxy.py it immediately reboots.
If something happens and i had to reconnect to the apm over serial in midair then a disaster would happen when i try to reconnect.
How can i achieve rebootless connection to the board?
Thanks...

peterbarker, 2018-01-19:
On Wed, 17 Jan 2018, mustaphos wrote:

> Can i disable the reboot when i try to connect to the APM board?
> I give power to the apm and it starts to work.
> But when i connect to it over serial using mavproxy.py it immediately reboots.
> If [...]

The board should not reboot when MAVProxy connects.  That's a very bad 
thing to have happen.

Exactly what sort of board are you using?

mustaphos, 2018-01-19:
I am using ardupilot mega board and connecting with USB port

peterbarker, 2018-01-19:
On Fri, 19 Jan 2018, mustaphos wrote:

> I am using ardupilot mega board and connecting with USB port

Ah.  Hmmm.  That board hasn't been supported by ArduPilot for a very long 
time.

Does connecting a different GCS have the same effect?

mustaphos, 2018-01-25:
I will try and let you know.
Thanks for your answer.