ArduPilot/MAVProxy #486: Crash while connecting to a simulator or APM via radio

whxru, 2018-01-18:
Firstly I built MAVProxy according to [Windows Development Environment](http://ardupilot.github.io/MAVProxy/html/development/mavdevenvwindows.html) and started a simulator:
```shell
dronekit-sitl copter-3.3 --home=31.8872318,118.8193952,5,353
```
Then I try to connect to it via MAVProxy:
```shell
py -2 mavproxy.py --master=tcp:127.0.0.1:5760 --moddebug=3
```
An exception was threw:
```
Connect tcp:127.0.0.1:5760 source_system=255
Log Directory:
Telemetry log: mav.tlog
MAV> fence breach
GPS lock at 0 meters
online system 1
STABILIZE> Mode STABILIZE
Exception in thread main_loop:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "mavproxy.py", line 783, in main_loop
    master.wait_heartbeat()
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 383, in wait_heartbeat
    return self.recv_match(type='HEARTBEAT', blocking=blocking)
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 344, in recv_match
    m = self.recv_msg()
  File "C:\Python27\lib\site-packages\pymavlink\mavutil.py", line 318, in recv_msg
    msg = self.mav.parse_char(s)
  File "C:\Python27\lib\site-packages\pymavlink\dialects\v10\ardupilotmega.py", line 9257, in parse_char
    self.__callbacks(m)
  File "C:\Python27\lib\site-packages\pymavlink\dialects\v10\ardupilotmega.py", line 9234, in __callbacks
    self.callback(msg, *self.callback_args, **self.callback_kwargs)
  File "build\bdist.win32\egg\MAVProxy\modules\mavproxy_link.py", line 375, in master_callback
    mavutil.mavlink.MAV_TYPE_DODECAROTOR]:
AttributeError: 'module' object has no attribute 'MAV_TYPE_DODECAROTOR'
```

Even though I tried to connect to a real copter with an APM board on it via 3DR radio, it crashed as well.

---

* OS: Windows 10
* pip list:
    * backports.functools-lru-cache (1.4)
    * cycler (0.10.0)
    * distribute (0.7.3)
    * dronekit (2.9.1)
    * dronekit-sitl (3.2.0)
    * future (0.15.2)
    * matplotlib (2.1.1)
    * mavproxy (1.6.2)
    * monotonic (1.2)
    * numpy (1.14.0)
    * opencv-python (3.4.0.12)
    * packaging (14.2)
    * pip (9.0.1)
    * psutil (5.4.1)
    * pygame (1.9.3)
    * PyInstaller (2.1)
    * pymavlink (2.0.6)
    * pyparsing (2.2.0)
    * pyreadline (2.1)
    * pyserial (3.4)
    * python-dateutil (2.6.1)
    * pytz (2017.3)
    * setuptools (19.2)
    * six (1.11.0)
    * wxPython (4.0.0b2)

SamuelDudley, 2018-01-18:
Hi, please update your pymavlink installation. The new release is apparently fixed as per https://github.com/ArduPilot/MAVProxy/issues/467

stephendade, 2018-01-18:
I've added in a patch to ensure the user is running the correct version of pymavlink.

whxru, 2018-01-20:
@SamuelDudley Thanks a lot, now the MAVProxy works correctly.

closed 2018-01-20 by whxru

hamidysf, 2018-06-28:
how to update pymavlink? but after I update pymavlink via pip, dronekit cannot run well ,

peterbarker, 2018-07-04:
On Thu, 28 Jun 2018, hamidysf wrote:

> how to update pymavlink? but after I update pymavlink via pip, dronekit cannot run well ,

Use dronekit-python from the github source.

nachod30, 2018-07-25:
I did "pip install pymavlink" and it works!!