ArduPilot/MAVProxy #494: ArduCopter: In mavproxy Auto mode just lands on newly built SITL setup

piningforfjords, 2018-03-07:
Completely built up an SITL setup today and found starting in GUIDED mode I could takeoff but the vehicle would land as soon as moe auto was entered
using the commands

wp load way.txt                 # flightplan I was trying to run shown below
arm throttle
takeoff 80                           # all works  great

mode auto                          # SITL just lands - vehicle makes slight progress towards wp1

flight plan looks fine on the map view, battery is at 90% 
flight plan:
QGC WPL 110
1	0	0	16	0.0	0.0	0.0	0.0	-35.362316	149.165146	80.0	1
2	0	0	16	0.0	0.0	0.0	0.0	-35.362297	149.165421	80.0	1
3	0	0	16	0.0	0.0	0.0	0.0	-35.363014	149.165497	80.0	1
4	0	0	16	0.0	0.0	0.0	0.0	-35.362991	149.165771	80.0	1
5	0	0	16	0.0	0.0	0.0	0.0	-35.362278	149.165695	80.0	1
6	0	0	16	0.0	0.0	0.0	0.0	-35.362259	149.165985	80.0	1
7	0	0	16	0.0	0.0	0.0	0.0	-35.362972	149.166061	80.0	1
8	0	0	16	0.0	0.0	0.0	0.0	-35.363033	149.165222	80.0	1
9	0	0	16	0.0	0.0	0.0	0.0	-35.363033	149.165222	10.0	1
10	0	0	20	0.0	0.0	0.0	0.0	0.0	0.0	0.0	1

peterbarker, 2018-03-07:
Your altitude frames are absolute.  That's no so much landing as doing a 
controlled flight into terrain.

piningforfjords, 2018-03-07:
Thanks.  Should have gone to the forum with this but the behavior changed from the previous build I was using.  The documentation for ardupilot is amazing but this is not covered in too many places.  Certainly no where searching the terms absolute, relative or altitude.  This is a recent change to ArduCopter as it didn't handle absolute altitude before 3.4.  I could see this burning people who tried to use it at the wrong place above sea level.   How you specify relative vs absolute in your flight plan:
QGC WPL 120
///// 0 in third position means absolute
1   0   0   16  5   0   0   1   lat   lon   hgt   1
///// 3 in third position means relative
1   0   3   16  5   0   0   1   lat   lon   hgt   1

  Mission planner doesn't seem to even populate the 3 column until it saves the flight plan and it leaves the column without a title.  Thanks again PB.

closed 2018-03-07 by piningforfjords