ArduPilot/MAVProxy #497: goto option in MAVProxy map not giving any result

Aashika-Chitagupi, 2018-03-23:
I am new to SITL and was working on MAVProxy triying to connect it with QGC on a windows system. When I tried to change the location on the map using the goto option it flashes the specified location and comes back to the original defult location. Please help

stephendade, 2018-03-23:
You can get MAVProxy support over in the ardupilot forums: https://discuss.ardupilot.org/c/ground-control-software/mavproxy

Aashika-Chitagupi, 2018-03-23:
Ok I will look into it. Thank you for replying.

peterbarker, 2018-03-23:
On Fri, 23 Mar 2018, Aashika-Chitagupi wrote:

> I am new to SITL and was working on MAVProxy triying to connect it with QGC
> on a windows system. When I tried to change the location on the map using
> the goto option it flashes the specified location and comes back to the
> original defult location. Please help

It is probably following the aircraft position.  ctrl-f toggles that 
I think.

Aashika-Chitagupi, 2018-03-24:
Thank you all I was able to solve the problem.

Aashika-Chitagupi, 2018-03-24:
I am able to go to a specific location now but the copter still stands stationed at the default location. How can i change that?