ArduPilot/MAVProxy #498: Joystick module fills terminal with "SDL_JoystickGetAxis value:"

AndKe, 2018-03-24:
(current master, on Linux)
Is there a  way to prevent this ?

suz1, 2018-04-04:
OK I have found this is because of SDL (a game library) printing debug messages, some sites are saying this is a compile time option for SDL, with debug options leading to printing this message. I am looking for a workaround (either piping this to dev null or recompiling the SDL module not to do this) as I also would like this functionality - it messes up SITL testing with a joystick as the MAVProxy window (though still functional) is spammed. I've tried looking through the code, but it appears that although there seems an aspiration to have verbose and debug modes, these switches ('joystick set debug, joystick set verbose') haven't been implemented yet. If I can find a workaround I will post it here.

peterbarker, 2018-04-04:
On Wed, 4 Apr 2018, Suzanna Lucarotti wrote:

> Ok solved. This problem is caused by a bug in pygame. You need at least

Thanks!  This bug annoys me any time I deal with the joystick stuff.

AndKe, 2018-08-25:
it seems so close, yet fail to install ...
```
$ pip install --upgrade pygame
Collecting pygame
  Downloading https://files.pythonhosted.org/packages/26/38/947e70c170e03f0e0aee61ae2e2157278d6146447d2b8cc8c46be5986b3e/pygame-1.9.4-cp27-cp27mu-manylinux1_x86_64.whl (12.0MB)
    100% |████████████████████████████████| 12.0MB 1.6MB/s 
Installing collected packages: pygame
  Found existing installation: pygame 1.9.1release
Cannot uninstall 'pygame'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.

```