ArduPilot/MAVProxy #499: how to handle it ?  version : python3.5

Zhenbin24, 2018-04-11:
- Traceback (most recent call last):
-   File "MAVExplorer.py", line 21, in <module>
-     from MAVProxy.modules.lib import wxconsole
-   File "/home/zhen24/Desktop/Exercise/lib/python3.5/site-packages/MAVProxy/modules/lib/wxconsole.py", line 7, in <module>
-     import textconsole, sys, time
- ImportError: No module named 'textconsole'

SamuelDudley, 2018-04-11:
MAVProxy and MAVExplorer do not officially support python 3.x at this time. My recommendation for you is to use python 2 unless you want to spend the time to get it working. 

The closest thing we have to python 3.x support is the branch in #420 (which as the title reads is a bit of a hack).

Zhenbin24, 2018-04-12:
@SamuelDudley thanks ！

monkeypants, 2018-04-12:
Can you explain what you mean a bit more please @Zhenbin24

Zhenbin24, 2018-04-12:
@monkeypants Nesting an image into a web page，like this
[image]
You can look at this：https://review.px4.io.

SamuelDudley, 2018-04-12:
@Zhenbin24 Am I correct in thinking that you want to write a web based graph display similar to https://review.px4.io but using MAVExplorer for processing the data logs?

If my assumption is correct, what you are describing (in my opinion) is a non trivial project that is separate to the core functionality of MAVExplorer. Perhaps you can review the source for that project on github and work from there: https://github.com/PX4/flight_review.

I personalty don't think raising an issue in this repository this is the correct way to be asking for support of this nature. Perhaps head over to the MAVProxy gitter channel and ask there.

Zhenbin24, 2018-04-12:
@SamuelDudley https://github.com/PX4/flight_review requires a ulog file, but I don't have it, only the bin file, and your project works fine, so I want to show the results of your project file on the web page and change the flight_review project to what you want Features (upload, analysis, upload record and click record display .....).

monkeypants, 2018-04-12:
are you proposing a new project, or a modification to PX4/flight_review?

Zhenbin24, 2018-04-12:
@monkeypants In general, it is to replace the analytical process of this project with the original（PX4/flight_review） one.

monkeypants, 2018-04-12:
Do you mean flight_review is the original, and MAVProxy is something else? I think of it the other way around :)

I see you have some comments/questions in the other project in GitHub, and it sounds like you are trying to combine code from this project into that one. This project is GPL and that one is BSD, so it won't be possible to copy and paste code unless that project upgrade's it's licence. However, like @SamuelDudley suggests, if you might find some inspiration from this code.

closed 2018-04-15 by Zhenbin24