ArduPilot/MAVProxy #505: No file named filelist_python (windows)

karthikdesai, 2018-05-03:
After using the latest mavproxy installer, connecting to the SITL i receive this error
[image]
On investigating, it was from this line
https://github.com/ArduPilot/MAVProxy/blame/master/MAVProxy/modules/mavproxy_map/srtm.py#L86

https://github.com/ArduPilot/MAVProxy/blame/master/MAVProxy/modules/mavproxy_map/srtm.py#L82
This line creates the directory, and the #L86 gets the path to the filelist_python from a directory which just got created. 
Is there any way to fix this? 
The PC is behind a firewall which could cause some unforseeable issues!

stephendade, 2018-05-03:
That's a known issue - MAVProxy map won't load if there's no cache and no internet connection.

There's a patch at #277, waiting to be merged in.

karthikdesai, 2018-05-03:
@stephendade cool thanks! will merge this patch and see if it solves it

karthikdesai, 2018-05-03:
@stephendade Okay it is working! Thank you! I hope #277 will be merged in master soon ^^

karthikdesai, 2018-05-04:
Will keep this open till the PR is merged!