ArduPilot/MAVProxy #506: Log download is very slow

GekoCH, 2018-05-03:
I just updated to the newest GitHub Mavproxy version but still I have a strange behavior during log downloads via Mavproxy. (I use a linux machine)
The first 97% of the logile is downloaded very fast via `log download "nr"` 
However the last part takes ages and displaying the log download status reveals a lot of retires.
After the download here the message:
Finished downloading log32.bin (11093798 bytes 503 seconds, 22.0 kbyte/sec 2856 retries)

And during download I get a lot of artifact data (see image). Is this a bug or how can I improve the log download?

[image]

arsenixprime, 2018-08-31:
I also have this issue frequently. Was excited to see this thread, disappointed no other comments.

GekoCH, 2018-08-31:
yeah seems nobody is using this feature....

peterbarker, 2018-09-01:
On Thu, 30 Aug 2018, arsenixprime wrote:

> I also have this issue frequently. Was excited to see this thread,
> disappointed no other comments.

What do you consider "very slow"?

You should expect 250kB/s over USB on master.

itzmejawad, 2021-03-22:
@peterbarker 

I tried setting serial0_baud with 9600,115200,921600... but for all these baudrates the log download speed is 150kbps. Unable to achieve 250kbps. Need ur help. Pixhawk 2.1 connected to companion computer. I am downloading via mavproxy