ArduPilot/MAVProxy #532: MAVExplorer not recognizing voltage data

laurenweinstein1, 2018-08-18:
MAVExplorer/MAVProxy seems to not be recognizing voltage data for newer (3.6?) ArduCopter versions. Apparently this was changed from CURR.Volt to BAT.Volt. So now on newer dataflash logs this battery related data doesn't appear to be recognized by MAVExplorer at all.

peterbarker, 2018-08-19:
On Sat, 18 Aug 2018, Lauren Weinstein wrote:

> 
> MAVExplorer/MAVProxy seems to not be recognizing voltage data for newer
> (3.6?) ArduCopter versions. Apparently this was changed from CURR.Volt to
> BAT.Volt. So now on newer dataflash logs this battery related data doesn't
> appear to be recognized by MAVExplorer at all.

mavproxy itself doesn't really have much to do with dataflash logs, which 
is where the CURR and BAT messages exist.  Is MAVProxy's GCS functionality 
impacted?

MAVExplorer is probably lacking canned graphs for the new fields - it's 
tricky because of the instance stuff going on in new messages, but BAT 
shouldn't be a problem.  Correcting this should be a matter of adding some 
entries to the pre-canned graph XML file.

laurenweinstein1, 2018-08-19:
I've only tested from the MAVExplorer side. I did dig around a little
and didn't see where to make the necessary change. Thanks!

--Lauren--
Lauren Weinstein (lauren@vortex.com): https://www.vortex.com/lauren 
Lauren's Blog: https://lauren.vortex.com
Google Issues Mailing List: https://vortex.com/google-issues
Founder: Network Neutrality Squad: https://www.nnsquad.org 
         PRIVACY Forum: https://www.vortex.com/privacy-info
Co-Founder: People For Internet Responsibility: https://www.pfir.org/pfir-info
Member: ACM Committee on Computers and Public Policy
Google+: https://google.com/+LaurenWeinstein
Twitter: https://twitter.com/laurenweinstein
Tel: +1 (818) 225-2800

peterbarker, 2018-08-19:
> I've only tested from the MAVExplorer side. I did dig around a little
> and didn't see where to make the necessary change. Thanks!

https://github.com/ArduPilot/MAVProxy/tree/master/MAVProxy/tools/graphs

Peter

> 
> 
> 
> >
> > >
> > > MAVExplorer/MAVProxy seems to not be recognizing voltage data for
> newer
> > > (3.6?) ArduCopter versions. Apparently this was changed from CURR.Volt
> to
> > > BAT.Volt. So now on newer dataflash logs this battery related data
> doesn't
> > > appear to be recognized by MAVExplorer at all.
> >
> > mavproxy itself doesn't really have much to do with dataflash logs,
> which
> > is where the CURR and BAT messages exist. Is MAVProxy's GCS
> functionality
> > impacted?
> >
> > MAVExplorer is probably lacking canned graphs for the new fields - it's
> > tricky because of the instance stuff going on in new messages, but BAT
> > shouldn't be a problem. Correcting this should be a matter of adding
> some
> > entries to the pre-canned graph XML file.
> >
> >
> > https://github.com/ArduPilot/MAVProxy/issues/532#issuecomment-414164466
> 
> 
> 
> 
> --Lauren--
> Lauren Weinstein (lauren@vortex.com): https://www.vortex.com/lauren
> Lauren's Blog: https://lauren.vortex.com
> Google Issues Mailing List: https://vortex.com/google-issues
> Founder: Network Neutrality Squad: https://www.nnsquad.org
> PRIVACY Forum: https://www.vortex.com/privacy-info
> Co-Founder: People For Internet Responsibility:
> https://www.pfir.org/pfir-info
> Member: ACM Committee on Computers and Public Policy
> Google+: https://google.com/+LaurenWeinstein
> Twitter: https://twitter.com/laurenweinstein
> Tel: +1 (818) 225-2800
> 
> thread.[AGv_4bki6-q0yF-lgj8ayOff2qoWFwVVks5uSfj4gaJpZM4WCpCh.gif]

Peter Barker                          |   Programmer,Sysadmin,Geek.
pbarker@barker.dropbear.id.au	      |   You need a bigger hammer.
:: It's a hack! Expect underscores! - Nigel Williams