ArduPilot/MAVProxy #543: Unable to wait for the incomming connections on --master due to unexpected keyword argument 'timeout'

brodnev, 2018-09-09:
Both --master=tcpin and --master=udpin returns "timeout" after 2 sec
pi64 ~ # mavproxy.py --master=tcpin:0.0.0.0:14550 --baudrate 57600 --out=udp:192.168.10.59:14555
Connect tcpin:0.0.0.0:14550 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from tcpin:0.0.0.0:14550
 MAV> Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python-exec/python2.7/mavproxy.py", line 798, in main_loop
    master.wait_heartbeat(timeout=0.1)
TypeError: wait_heartbeat() got an unexpected keyword argument 'timeout'

pi64 ~ # mavproxy.py --master=udpin:0.0.0.0:14550 --baudrate 57600 --out=udp:192.168.10.59:14555
Connect udpin:0.0.0.0:14550 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from 0.0.0.0:14550
 MAV> Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib64/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib64/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python-exec/python2.7/mavproxy.py", line 798, in main_loop
    master.wait_heartbeat(timeout=0.1)
TypeError: wait_heartbeat() got an unexpected keyword argument 'timeout'

brodnev, 2018-09-09:
P.S. sources has been downloaded today, 09.09.2018

karthikdesai, 2018-09-18:
https://github.com/ArduPilot/MAVProxy/pull/544 try it with this PR @brodnev

umer936, 2018-09-30:
Had this issue as well since I needed to be back a pymavlink version. Did the PR manually and worked fine. 

More details on my situation (ie why I needed pymavlink==2.2.7) here: https://github.com/ArduPilot/pymavlink/issues/222

peterbarker, 2018-10-04:
On Sun, 30 Sep 2018, Umer Salman wrote:

> Had this issue as well since I needed to be back a pymavlink version. Did
> the PR manually and worked fine.

tridge has updated the minimum requirements for MAVProxy now, so I've 
killed that PR now.

amilcarlucas, 2018-10-17:
Works in the latest release. Closing it now. Please reopen if the problem persists in the latest release.

closed 2018-10-17 by amilcarlucas

tobiasfriden, 2018-11-02:
Hello,

I still have problems with this using the latest release.
Trying to run ardupilot sitl in a Docker container.
If i fix mavproxy and pymavlink versions:
```
pip install mavproxy==1.6.2
pip install pymavlink==2.2.8
```
Everything works but running
```
pip install --upgrade mavproxy
```
which results in mavproxy v1.7.7 and pymavlink v2.2.21 I get the above error.

amilcarlucas, 2018-11-02:
please do:
pip install pymavlink==2.2.21
pip install mavproxy==1.7.7

That should solve all the problems.

tobiasfriden, 2018-11-02:
@amilcarlucas unfortunately leads to the same issue. In which way would this be different from 
```
pip install --upgrade mavproxy
```
as it leads to the same installed versions?

amilcarlucas, 2018-11-02:
Installing pymavlink first might be helpful.
Those versions run fine here, so I think you have outdated versions somewhere in your system, and those are getting somehow used.

galdalali, 2018-11-25:
Same issue here when running SITL with ArduCopter-3.5.0 after trying to rebuild Pymavlink in order to create custom mavlink massage:

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/bin/mavproxy.py", line 811, in main_loop
    master.wait_heartbeat(timeout=0.1)
TypeError: wait_heartbeat() got an unexpected keyword argument 'timeout'
```

amilcarlucas, 2018-11-26:
It is due again to the same issue, mixing old and new versions of some modules.
Please use:
- ArduCopter 3.6.2
pip install pymavlink==2.2.21
pip install mavproxy==1.7.7

galdalali, 2018-11-26:
I can not. I'm trying to add a custom mavlink message to ArduCopter-3.5.0 and since MavProxy in to really a proxy I need to compile pymavlink again.

galdalali, 2018-11-26:
I added a custom mavlink massge to ardupilotmega.xml and it works well with Ardupilot hardwere connected via serial to a mavlink client.

galdalali, 2018-11-26:
I added a custom mavlink massage to ardupilotmega.xml and it works well with Ardupilot hardware connected via serial to a mavlink client. 
Now I want to do the same with SITL, so I recompiled with SITL and (runing Mavproxy through sim_vehicle.py) and I can not see the massage.

galdalali, 2018-11-26:
From a small research I discovered that MavProxy creating his headers VIA Pymavlink massage_definitions so I need to recompile Pymavlink again with an updated ardupilotmega.xml right?

amilcarlucas, 2018-11-26:
yes

galdalali, 2018-11-26:
So this is what's happened:

> Same issue here when running SITL (sim_vhicle.py) with ArduCopter-3.5.0 after trying to rebuild Pymavlink in order to create custom mavlink massage:
> 
> ```
> Traceback (most recent call last):
>   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
>     self.run()
>   File "/usr/lib/python2.7/threading.py", line 754, in run
>     self.__target(*self.__args, **self.__kwargs)
>   File "/usr/local/bin/mavproxy.py", line 811, in main_loop
>     master.wait_heartbeat(timeout=0.1)

> TypeError: wait_heartbeat() got an unexpected keyword argument 'timeout'
> ```

amilcarlucas, 2018-11-26:
Old mavproxy code with new pymavlink code.

galdalali, 2018-11-26:
Ok now I understand, with this versions:

mavproxy==1.6.2
pymavlink==2.2.8
ArduCopter==3.5.0

Everything works fine, so in the sources of pymavlink related to this version of ArduPilot, I changed to "version = '2.2.8'"  in setup.py, and now it'ws works well

amilcarlucas, 2018-11-26:
It would also work well if you had un-installed all SW and installed:
mavproxy==1.7.7
pymavlink==2.2.21
ArduCopter==3.5.0

galdalali, 2018-11-26:
Nope, for some reason that gives me the timeout issue..

maroquio, 2018-12-21:
I got this same error using pymavlink 2.3.2. I downgraded to pymavlink 2.3.1 and everything worked fine again. Cheers!