ArduPilot/MAVProxy #564: Load map issue on Mac OS

stodi1, 2018-10-23:
OS: MacOS Sierra 10.12.6
Python: 3.7
Using the latest release of MAVProxy 1.7.6

When I try staring the map module using: mavproxy.py --master=127.0.0.1:14550 --cmd="param load init.parm; modu load map;"

I have some exceptions:
/usr/local/bin/mavproxy.py:19: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  from imp import reload
/usr/local/lib/python3.7/site-packages/billiard/context.py:261: RuntimeWarning: force_execv is not supported as the billiard C extension is not installed
  warnings.warn(RuntimeWarning(W_NO_EXECV))
Connect 127.0.0.1:14550 source_system=255
Process Process-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 103, in loadFileList
    data = open(self.filelist_file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/Users/abdou/.tilecache/SRTM/filelist_python'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 327, in _bootstrap
    self.run()
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 166, in createFileListHTTP
    parser.feed(data)
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/html/parser.py", line 110, in feed
    self.rawdata = self.rawdata + data
TypeError: can only concatenate str (not "bytes") to str

then some other errors:
Failed to open file 'init.parm'
Loaded module map
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from 127.0.0.1:14550
MAV> Process Process-2:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 103, in loadFileList
    data = open(self.filelist_file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/Users/abdou/.tilecache/SRTM/filelist_python'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 327, in _bootstrap
    self.run()
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 166, in createFileListHTTP
    parser.feed(data)
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/html/parser.py", line 110, in feed
    self.rawdata = self.rawdata + data
TypeError: can only concatenate str (not "bytes") to str
Process Process-3:1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 103, in loadFileList
    data = open(self.filelist_file, 'rb')
FileNotFoundError: [Errno 2] No such file or directory: '/Users/abdou/.tilecache/SRTM/filelist_python'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 327, in _bootstrap
    self.run()
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 166, in createFileListHTTP
    parser.feed(data)
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/html/parser.py", line 110, in feed
    self.rawdata = self.rawdata + data
TypeError: can only concatenate str (not "bytes") to str
Process Process-3:2:
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 327, in _bootstrap
    self.run()
  File "/usr/local/lib/python3.7/site-packages/billiard/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 166, in createFileListHTTP
    parser.feed(data)
  File "/usr/local/Cellar/python/3.7.0/Frameworks/Python.framework/Versions/3.7/lib/python3.7/html/parser.py", line 110, in feed
    self.rawdata = self.rawdata + data
TypeError: can only concatenate str (not "bytes") to str

The map loads and quit fastly with a mac crash error : "python quit unexpectedly".

I don't know where this issue came from.
Regards.

blueForestIcarus, 2019-02-15:
I also have this issue, but on Arch Linux - python 3.7.2

xlla, 2019-03-06:
same issue here,
OS: MacOS Sierra 10.12.6
Python: 3.7
mavproxy 1.8.5

xlla, 2019-03-06:
in srtm.py, add 
 data = data.decode("utf-8")

now it start downloading....

peterbarker, 2019-03-11:
On Wed, 6 Mar 2019, xlla wrote:

> in srtm.py, add
> data = data.decode("utf-8")
> 
> now it start downloading....

Could you create a PR, please?

stephendade, 2019-03-11:
> Could you create a PR, please?

And as luck would have it, this is included in my "general Python3 fixes" patch #603

srmainwaring, 2023-04-25:
Closing as no recent activity and map is working on macOS Monterey.

closed 2023-04-25 by srmainwaring