ArduPilot/MAVProxy #565: pip install MAVProxy not working on OSX Mojave

jmachuca77, 2018-10-23:
I tried installing MAVProxy with pip, on a fresh install of Mac OS X Mojave. Pip was not installed by default so I had to install it using this command:
```sudo easy_install pip```

once that was done I ran:
```sudo pip install mavproxy```

and ran into an issue with numpy, seems bumpy 1.8.0rc1 is installed by default, but python-opencv requires >=1.11.1, however when trying to update numpy I get an error:

```Cannot uninstall 'numpy'. It is a distutils installed project and thus we cannot accurately determine which files belong to it which would lead to only a partial uninstall.```

Attached is the terminal dump

[Terminal Saved Output.txt](https://github.com/ArduPilot/MAVProxy/files/2508361/Terminal.Saved.Output.txt)

gniewus, 2019-01-29:
Update! Map module works, but only in FullScreen mode.

gniewus, 2019-02-01:
It comes with Python 2.7, but I'm using the Anaconda distribution.

monkeypants, 2019-02-05:
What version of python in your Anaconda?

srmainwaring, 2023-04-25:
Hi @jmachuca77, going through old MAVProxy / macOS tickets. I think the macOS install issues are largely resolved for macOS Monterey on Intels and M1s. I'll close due to no recent activity, but please reopen if the issue persists on newer versions.

closed 2023-04-25 by srmainwaring