ArduPilot/MAVProxy #568: pkg_resources.ResolutionError: Script 'scripts/mavproxy.py'...

bobypierre, 2018-11-05:
Hi,

I have this issue when I wish to lauch MAVProxy. I'm a beginner with MAVProxy !:) In order to install MAVProxy (on ubuntu 18.) I follow this: http://ardupilot.github.io/MAVProxy/html/getting_started/download_and_installation.html#linux
and without problem until this command...:  mavproxy.py .........
Here is the line return on my terminal:
''
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 4, in <module>
    __import__('pkg_resources').run_script('MAVProxy==1.7.7', 'mavproxy.py')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1429, in run_script
    .format(**locals()),
pkg_resources.ResolutionError: Script 'scripts/mavproxy.py' not found in metadata at '/home/boby/.local/lib/python2.7/site-packages/MAVProxy-1.7.7.dist-info'
''

I'll be happy if someone can help!

peterbarker, 2018-11-05:
On Mon, 5 Nov 2018, bobypierre wrote:

Please run "pip --version" and paste the results in.

bobypierre, 2018-11-06:
Hi,
Thanks for your feedback!

pip -- version return:
pip 9.0.1 from /usr/lib/python2.7/dist-packages (python 2.7)

There is also python3, python 3.6 and 3.7 installed by default on my system (I'm seeing in usr/lib). Like MAVProxy can only run python 2.7, is the last versions could bring some problem? 

have a good day,
Pierre

peterbarker, 2018-11-06:
Could you try "sudo pip uninstall MAVProxy` (repeatedly, until it stop saying it has removed something), then `pip install --user MAVProxy` , please?

bobypierre, 2018-11-06:
Yes I did uninstall until this message: 
"Cannot uninstall requirement MAVProxy, not installed"
and did 
"pip install --user MAVProxy"

terminal saying: 
"Collecting MAVProxy
Collecting pymavlink>=2.2.19 (from MAVProxy)
Collecting pyserial>=3.0 (from MAVProxy)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting future (from pymavlink>=2.2.19->MAVProxy)
Collecting lxml (from pymavlink>=2.2.19->MAVProxy)
  Using cached https://files.pythonhosted.org/packages/e5/14/f4343239f955442da9da1919a99f7311bc5627522741bada61b2349c8def/lxml-4.2.5-cp27-cp27mu-manylinux1_x86_64.whl
Installing collected packages: future, lxml, pymavlink, pyserial, MAVProxy
Successfully installed MAVProxy-1.7.7 future-0.17.1 lxml-4.2.5 pymavlink-2.2.21 pyserial-3.4"

Then I tested "mavproxy.py" and the problem remaining: 
boby@boby-K53SD:~$ mavproxy.py
"Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 4, in <module>
    __import__('pkg_resources').run_script('MAVProxy==1.7.7', 'mavproxy.py')
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 658, in run_script
    self.require(requires)[0].run_script(script_name, ns)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 1429, in run_script
    .format(**locals()),
pkg_resources.ResolutionError: Script 'scripts/mavproxy.py' not found in metadata at '/home/boby/.local/lib/python2.7/site-packages/MAVProxy-1.7.7.dist-info'"

bobypierre, 2018-11-06:
sorry but I'm asking me one idiot question maybe, I have previously installed apm planner for test and I hope that it not disturb MAVProxy

bobypierre, 2018-11-06:
I also note my site-packages repertory (usr/local/lib/python2.7/site-packages) is empty. Is it normal?

bobypierre, 2018-11-06:
RESOLVED ?

Hello,
I decided to format my computer in order to work in good bases and reinstall Linux... Then I have reinstall all modules with MAVProxy. 
Now, when I lauch "mavproxy.py" without other options, the terminal show me the auto detected serial port.
I think is better now? 
The most negative thing is I'm not understand all!

Thanks a lot and congratulation for your work..

peterbarker, 2018-11-06:
On Tue, 6 Nov 2018, bobypierre wrote:

> I decided to format my computer in order to work in good bases and reinstall

Wow, that's a pretty big hammer to fix this problem!

From your previous email I could see that there was a 
/usr/local/bin/mavproxy.py which should not longer have existed; the "pip 
uninstall" steps should have gotten rid of all of that.  Had the situation 
still existed we could have progressed :-)

Did you ever install mavproxy from source?

peterbarker, 2018-11-08:
On Wed, 7 Nov 2018, bobypierre wrote:

> I have a new problem to continue! (bug during the compilation of the module SITL, and with magical module (...GLSL3.30 is not supported...) to calibrate
> the magnetometer).
> I will progress now and should I post this may problems in a new issue?

Please do.