ArduPilot/MAVProxy #569: "magical_ui" ------>Exception: Error on compiling fragment shader:

bobypierre, 2018-11-08:
Hi,

When I execute the command "magical_ui", the calibrate's window open, but this errors notes below appear in loop (if the mouse move along the window) and impossible to launch the calibration.

Phase:
1. mavproxy.py --master 0.0.0.0:14550 --aircraft Bebop2 --load console 
2. STABILIZE> module load magical
3. STABILIZE> magical_ui

NB: I have installed PyOpenGL before to can lauch magical_ui.

STABILIZE> Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
    self.CreateRenderer()
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxvehicle.py", line 79, in CreateRenderer
    background=(r / 255.0, g / 255.0, b / 255.0, 1),
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxvehicle.py", line 33, in __init__
    super(Renderer, self).__init__(background)
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 34, in __init__
    self.program.compile_and_link()
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/lib/opengl.py", line 601, in compile_and_link
    raise Exception("Error on compiling %s: %s" % (name, error_string))
Exception: Error on compiling fragment shader: 0:2(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 87, in OnPaint
    self.CreateRenderer()
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 143, in CreateRenderer
    background=(r / 255.0, g / 255.0, b / 255.0, 1),
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 35, in __init__
    super(Renderer, self).__init__(background)
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/glrenderer.py", line 34, in __init__
    self.program.compile_and_link()
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/lib/opengl.py", line 601, in compile_and_link
    raise Exception("Error on compiling %s: %s" % (name, error_string))
Exception: Error on compiling fragment shader: 0:2(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES..........................................
........................
.....................
.....................................
NameError: global name 'event' is not defined
Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 243, in OnMotion
    if hasattr(event, 'ButtonIsDown'):
NameError: global name 'event' is not defined
Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 243, in OnMotion
    if hasattr(event, 'ButtonIsDown'):
NameError: global name 'event' is not defined
Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 243, in OnMotion
    if hasattr(event, 'ButtonIsDown'):
NameError: global name 'event' is not defined
Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 243, in OnMotion
    if hasattr(event, 'ButtonIsDown'):......................
...........................................
......................................
.................................
.................................

bobypierre, 2018-11-08:
As well as, here is my output of sudo glxinfo | grep "OpenGL" (found on this forum: https://stackoverflow.com/questions/40572941/c-opengl-shading-version-error-glsl-x-is-not-supported-ubuntu-16-04)

OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Sandybridge Mobile 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

bobypierre, 2018-11-08:
Hi,
I'm afraid it's my graphic card which don't support...NVIDIA 610M...
I don't know

bobypierre, 2018-11-08:
!!!!!!!!!!!!!!!  some better news   !!!!
I think that I had two problem, but one is resolved ! 

I attempted to change the driver of my graphic card (it was a free driver, and I have installed an owner driver NVIDIA), Now I can see the calibration window entire and the copter move when I move my bebop. no more problem with GLSL 3.30... !!

On the other hand, when I "start" the calibration, the counter start too, but arrived to 0, no process.. and this error in loop in the Shell:   

STABILIZE> Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 243, in OnMotion
    if hasattr(event, 'ButtonIsDown'):
NameError: global name 'event' is not defined
Traceback (most recent call last):
  File "/home/boby/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_magical/wxgeodesicgrid.py", line 243, in OnMotion
    if hasattr(event, 'ButtonIsDown'):
NameError: global name 'event' is not defined
Traceback (most recent call last):

Have a good time !
Pierre

peterbarker, 2018-11-08:
On Thu, 8 Nov 2018, bobypierre wrote:

> NameError: global name 'event' is not defined

I've pushed a fix in MAVProxy's master branch for this.

Peter

bobypierre, 2018-11-09:
OK..Thank you.