ArduPilot/MAVProxy #598: PingRX ADS-B causes MAVProxy to lose flight mode mapping

RickReeser, 2019-03-01:
Making an issue from this discuss post:
https://discuss.ardupilot.org/t/pingrx-spamming-mavlink/38472/8?u=anubis

Basically, when I have a PingRX plugged in and ADSB_ENABLED = 1, MAVProxy doesn't know what flight modes are available. I get these errors when trying to change or list flight modes:

```
LOITER> loiter
LOITER> ERROR in command: argument of type 'NoneType' is not iterable
Unknown command 'loiter'
```
```
LOITER> mode
LOITER> No mode mapping available
```

My guess is that the Ping's heartbeats are being handled as flight controller heartbeats, and fails to get a flight mode list because the MAV_AUTOPILOT parameter (or whichever is used to determine available flight modes) is 0. I suppose Mavproxy should be able to discriminate heartbeats based on their component ID or type.

Peter Barker [added a provision to pymavutil](https://github.com/ArduPilot/pymavlink/pull/270) to do this, but I'm still experiencing flight mode problems after applying this fix.

peterbarker, 2019-03-01:
On Thu, 28 Feb 2019, RickReeser wrote:

Which pymavlink are you using?

RickReeser, 2019-03-01:
```
pip list | grep -i mav
pymavlink (2.3.4)
MAVProxy (1.8.5)
```
Also have your pymavutil PR.

peterbarker, 2019-03-13:
Could you supply a short tlog, please?

RickReeser, 2019-03-19:
[2019-03-19 12-18-49.tlog.zip](https://github.com/ArduPilot/MAVProxy/files/2985271/2019-03-19.12-18-49.tlog.zip)

Here's a tlog. It's not a .zip, just remove the .zip extension. It's just a bare Pixhawk 2 with a GPS and ADS-B on my bench. Let me know if that's enough.

RickReeser, 2019-03-21:
Just encountered another symptom. I flew the drone with the Ping plugged into my Pixhawk, but with `ADSB_ENABLE = 0`, which caused the mode errors to disappear. However, as soon as the drone armed, Mavproxy spammed these messages repeatedly, at about 1 Hz:
```Arming checks disabled                                                          |
DISARMED                                                                        |
ARMED                                                                           |
Arming checks disabled                                                          |
DISARMED                                                                        |
ARMED                                                                           |
Arming checks disabled                                                          |
DISARMED                                                                        |
height 0                                                                        |
ARMED                                                                           |
Arming checks disabled                                                          |
DISARMED                                                                        |
ARMED                                                                           |
Arming checks disabled                                                          |
DISARMED                                                                        |
ARMED                                                                           |
Arming checks disabled 
```

I assume this is just another symptom of the same problem. But maybe it will be helpful.