ArduPilot/MAVProxy #599: Mavproxy Crashing when I try to connect with Apm Planner

kaangoksal, 2019-03-03:
Whenever I try to connect with apm planner 2 to mavproxy, mavproxy crashes with the following error:

``` python

Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/bin/mavproxy.py", line 883, in main_loop
    process_mavlink(m)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/bin/mavproxy.py", line 625, in process_mavlink
    msgs = slave.mav.parse_buffer(buf)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/lib/python3.5/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 11821, in parse_buffer
    m = self.parse_char(s)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/lib/python3.5/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 11758, in parse_char
    m = self.__parse_char_legacy()
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/lib/python3.5/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 11808, in __parse_char_legacy
    m = self.decode(mbuf)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/lib/python3.5/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 11990, in decode
    tlist[i] = tlist[i].decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xdc in position 4: invalid continuation byte

```

I'm running the mavproxy like this

`venv/bin/mavproxy.py --master=/dev/serial0 --baud=57600 --out=tcpin:0.0.0.0:14550 --out=tcpin:127.0.0.1:17500`
I'm using it on python3

When I press connect with apm planner it crashes. I have no problems using dronekit with it

kaangoksal, 2019-03-03:
Also having this as well when I type help

``` python

STABILIZE> Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.5/threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/bin/mavproxy.py", line 840, in main_loop
    process_stdin(c)
  File "/home/pi/talos_c2c_governor_deployment/c2c/drone_governor_code/venv/bin/mavproxy.py", line 541, in process_stdin
    k.sort()
AttributeError: 'dict_keys' object has no attribute 'sort'

```

jo-m, 2019-03-04:
cannot reproduce the first issue, what version of APM planner is this?

kaangoksal, 2019-03-11:
My problem was resolved after I installed `sudo apt-get install python3-pip python3-dev gcc python3-lxml` I think it was a lxml issue

closed 2019-03-11 by kaangoksal

peterbarker, 2019-03-13:
On Mon, 11 Mar 2019, Kaan Göksal wrote:

> My problem was resolved after I installed sudo apt-get install python3-pip python3-dev gcc python3-lxml I think it was a lxml issue

How *did* you have things installed?

If you did have them installed from pipi we may have a problem...