ArduPilot/MAVProxy #600: MAVProxy crashing after connecting Mission Planner

jo-m, 2019-03-03:
I also saw this very similar issue: https://github.com/ArduPilot/MAVProxy/issues/599, but the error seems to occur in a different place. So if those two issues point to the same underlying problem in MAVProxy, this one can be closed.

Also, not sure if this bug belongs here or in pymavlink.

# Poblem/How to reproduce
1. Run SITL
2. Run MAVProxy `./venv/bin/mavproxy.py --master tcp:localhost:5760 --out=127.0.0.1:14551`
3. Connect Mission Planner on UDP port 14551

Will result in
```
Exception in thread main_loop:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "./venv/bin/mavproxy.py", line 883, in main_loop
    process_mavlink(m)
  File "./venv/bin/mavproxy.py", line 633, in process_mavlink
    mpstate.master().write(m.get_msgbuf())
  File "/home/joni/Documents/skyways-git/ship-gcs/venv/lib/python3.6/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 100, in get_msgbuf
    return bytearray(self._msgbuf)
TypeError: string argument without an encoding
```

I added the following debug lines in `ardupilotmega.py` on line 98:
```
import sys
sys.stderr.write(">>>>>>> %s %s\n" % (type(self._msgbuf), repr(self._msgbuf)))
```

which results in the following output right before the crash above:
```
>>>>>>> <class 'array.array'> array('B', [254, 22, 135, 1, 1, 193, 150, 188, 101, 60, 144, 154, 237, 59, 217, 0, 141, 59, 165, 164, 0, 59, 248, 30, 227, 59, 127, 3, 183, 249])
>>>>>>> <class 'array.array'> array('B', [254, 32, 136, 1, 1, 241, 13, 80, 29, 99, 0, 0, 0, 0, 164, 43, 188, 59, 163, 111, 193, 59, 139, 244, 29, 61, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 225, 114])
>>>>>>> <class 'array.array'> array('B', [254, 32, 136, 1, 1, 241, 13, 80, 29, 99, 0, 0, 0, 0, 164, 43, 188, 59, 163, 111, 193, 59, 139, 244, 29, 61, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 225, 114])
>>>>>>> <class 'str'> 's'
Exception in thread main_loop:
Traceback (most recent call last):
  ... (see above)
```

# Software Versions
```
MAVproxy: 1.8.5
Python:  3.6.7
Mission Planner: 1.3.62
```

kaangoksal, 2019-03-03:
I'm currently downgrading my mavproxy versions, and testing

kaangoksal, 2019-03-03:
can you type help on the mavproxy console?

jo-m, 2019-03-03:
yep getting the same error you got. But this seems unrelated to the encoding problem.

kaangoksal, 2019-03-03:
Can you install the following packages and let me know if your connection now work? 
`sudo apt-get install python3-pip python3-dev gcc python3-lxml`

jo-m, 2019-03-03:
I have all this stuff already installed.

Those two issues most probably are Python 2 -> 3 problems.
In Python 3 vs. 2, behaviour and default types for bytes and strings changed.
Also, in Python2 dict.keys() returned a list, but in Python3 it returns a dict_keys iter object which causes the error because it does not have the sort method.

jo-m, 2019-03-03:
@peterbarker @tridge I'd be happy to start working on a PR if this is not already fixed somewhere else.

stephendade, 2019-03-04:
For this and #599, the issue might be in pymavlink instead. What version of pymavlink are you running?

jo-m, 2019-03-04:
Pymavlink `2.3.4`.
But I think this is a problem with MAVProxy and not underlying pymavlink.

wx4cb, 2019-03-04:
i had the same problem... run it with python3 and it doesnt work. run it with python 2 it works perfectly. in fact runs great under ubuntu on WSL lol

it's definitively a python 3 issue as sort i beleive doesn't exist

jo-m, 2019-03-04:
created two PRs, one in pymavlink fixing the underlying problem and another one avoiding forwarding BAD_DATA messages (which was also involved in causing this bug).

jo-m, 2019-05-27:
the underlying bug in pymavlink has been fixed, this issue can be closed as soon as the pymavlink MAVProxy uses is bumped.

askarthikkumar, 2019-05-28:
I have updated my pymavlink to theatest release. How to change the version of pymavlink being used by mavproxy? Because the problem still persists even after updating pymavlink.

jo-m, 2019-05-28:
yes. There has been no release of pymavlink since the bug fix was introduced. As soon as pymavlink is updated (2.3.5 will be next I think), MAVproxy can be upgraded to use this version.

jo-m, 2019-05-31:
this should now be solved: https://github.com/ArduPilot/MAVProxy/commit/d1d127e3ae12e504b8ca497f346e5d26477e0b08

MAVProxy release 1.8.7, pymavlink 2.3.6.

closed 2019-05-31 by jo-m