ArduPilot/MAVProxy #601: MAVProxy MPState Error

peasant98, 2019-03-04:
The following command gives me this error:

`mavproxy.py --master=udp:127.0.0.1:14650`

**Result**:

Connect udp:127.0.0.1:14650 source_system=255
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 1084, in <module>
    load_module(m, quiet=True)
  File "/usr/local/bin/mavproxy.py", line 281, in load_module
    module = m.init(mpstate)
  File "/home/pi/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 433, in init
    return ParamModule(mpstate, **kwargs)
  File "/home/pi/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 363, in __init__
    self.check_new_target_system()
  File "/home/pi/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 410, in check_new_target_system
    self.add_new_target_system(sysid)
  File "/home/pi/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 383, in add_new_target_system
    if not sysid in self.mpstate.mav_param_by_sysid:
AttributeError: 'MPState' object has no attribute 'mav_param_by_sysid'

**Current Development Tools**:

I currently am using Python 2.7.13, and just got mavproxy today. I'm working with a Raspberry Pi as well.

SatishUpadhyaya, 2019-03-04:
I had the same issue!

stephendade, 2019-03-05:
I've tried this with Python 2.7.15 on my laptop, but don't get this issue. This was connecting to Arduplane SITL.

What vehicle type are you trying to connect to?

peasant98, 2019-03-05:
I'm currently trying to connect to Arducopter (real drone).

stephendade, 2019-03-06:
Looking at that error message again, it appears to be from an older version of MAVProxy. What version are you running? You can use ``mavproxy.py --version``

peasant98, 2019-03-08:
Sorry for the late response; I haven't been able to use the Pi; I will let you know what version I have soon.

adellusakula, 2019-08-17:
Hi, 
I have the same issue as @peasant98.
My python version is  2.7.6 and I updated my pymavlink version to 2.3.8 and I still have the same problem.
 Connect udp:127.0.0.1:14650 source_system=255
Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 1040, in <module>
    load_module(m, quiet=True)
  File "/usr/local/bin/mavproxy.py", line 275, in load_module
    module = m.init(mpstate)
  File "/home/viki/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 433, in init
    return ParamModule(mpstate, **kwargs)
  File "/home/viki/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 363, in __init__
    self.check_new_target_system()
  File "/home/viki/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 410, in check_new_target_system
    self.add_new_target_system(sysid)
  File "/home/viki/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_param.py", line 383, in add_new_target_system
    if not sysid in self.mpstate.mav_param_by_sysid:
AttributeError: 'MPState' object has no attribute 'mav_param_by_sysid'

peterbarker, 2019-08-23:
On Sat, 17 Aug 2019, adellusakula wrote:

> I have the same issue as @peasant98.
> My python version is 2.7.6 and I updated my pymavlink version to 2.3.8 and I still have the same problem.

This problem would appear to be confined to mavproxy itself, but it was 
definitely worth a punt on upgrading pymavlink.

This could happen if the mavproxy binary being run is at odds with the 
libraries being sourced.  This is the only scenario I've come up with. 
Please try purging old versions of MAVProxy and install again.

That means (assuming you're just using pip to install things), repeatedly 
running "pip2 remove MAVProxy" and "pip3 remove MAVProxy" until they both 
report none are installed.  Then install again ("pip install MAVProxy")

linuxguy123, 2020-01-06:
"That means (assuming you're just using pip to install things), repeatedly
running "pip2 remove MAVProxy" and "pip3 remove MAVProxy" until they both
report none are installed.  Then install again ("pip install MAVProxy")"

Thank you for posting this because I had the "AttributeError: 'MPState' object has no attribute 'mav_param_by_sysid'" error and fixed it by following your instructions.