ArduPilot/MAVProxy #608: Can't install on python2, can't run on python3

Bryan-Tam, 2019-03-27:
To my understanding, MAVProxy only runs on python2, however I am unable to install it using the command `sudo pip install mavproxy`. This ends up retrying multiple times and throws the SSL certificate error that implies that my pip is outdated. Pip can install other modules such as numpy, so I am unsure why MAVproxy is the problem. I can however install it with pip3 (for python 3) but it throws errors about missing directories and is unable to convert bytes to string. What might be the issue that prevents me from installing on python2 properly (unless it now works for python3)?

stephendade, 2019-03-27:
MAVProxy _should_ work on Python3. You may need to install from the github repo to get the latest fixes though.

Bryan-Tam, 2019-03-27:
That must be it, since I installed on python 3 through pip. Thanks.

closed 2019-03-27 by Bryan-Tam