ArduPilot/MAVProxy #609: Not drawing the mission when list is run

gmorph, 2019-04-06:
I've rebased to the latest ardupilot master and MAVProxy and I'm running XUbuntu 18.10.  When I do  "wp list" or a list via the map menu it fetches the waypoints just fine but doesn't draw them.  It only draw's WP1 and then nothing else.  No lines - nothing.  I only get WP1 present on the map.  When I draw a mission the "purple" lines appear as I'm drawing but once completed it has the same behaviour - only WP1 shows up.

peterbarker, 2019-04-06:
yyOn Sat, 6 Apr 2019, Grant Morphett wrote:

> I've rebased to the latest ardupilot master and MAVProxy and I'm running XUbuntu 18.10. When I do "wp list" or a list via the map menu it fetches the
> waypoints just fine but doesn't draw them. It [...]

I fixed this a week ago with 0fa49415a207937396d16d301cb7ed14c9f4db2e in 
pymavlink - please try updating your pymavlink install and trying again.

I don't think we've done a pymavlink release with this problem in it - but 
I could be wrong - so you must be working from a git repo to have this 
problem...

gmorph, 2019-04-06:
Hi Peter.  Yeah I was installing pymavlink from ardupilot master modules/mavlink directory.  I can't see the hash you have provided so perhaps ardupilot module hasn't been updated?

closed 2019-04-06 by gmorph