ArduPilot/MAVProxy #620: Ardupilot SITL joystick looking for mavproxy_joystick.py that isn't there anymore

UASOP, 2019-04-30:
Greetings all! 
I am attempting to add the GREAT PLANES InterLink-X controller to the list of joysticks for SITL. I am following the instructions for the new YAML file and I am making sure the match name is set properly along with the axis and buttons. I keep running into the issue that the joystick module is looking for the mavproxy_joystick.py script in the mavproxy modules folder and it isn't there anymore as part of the master git repo. https://github.com/ArduPilot/MAVProxy/tree/master/MAVProxy/modules

If it keeps looking for this file and it is not there, how is it supposed to load any of the referencable joysticks within the joysticks folder? I tried taking the mavproxy_joystick.py script from another source and plugging it into the master mavproxy module folder on my computer, but that did not make any change to the result. I have posted what the terminal is outputting to me below. 

 STABILIZE> MAVProxy.modules.mavproxy_joystick: Found joystick (GREAT PLANES InterLink-X)
MAVProxy.modules.mavproxy_joystick: Failed to find matching joystick.
Loaded module joystick

Any help would be greatly appreciated.

peterbarker, 2019-04-30:
You may need to install MAVProxy for it to pick up your new yaml definition.

You state MAVProxy is unable to find `mavproxy_joystick.py`?  What evidence do you have of that?  Your output from MAVProxy learly shows the module running, so I'm a little confused :-)

That particular file doesn't actually exist any more; there's now an `__init__.py` within an appropiately named folder - it's just an alternate structure that achieves the same goal but can result in a neater code layout.

Suggest you could `set modedebug 3`, `module unload joystick`, `module load joystick` to get more diagnostics.  We should add more debug into the probe function at higher debug levels - this still won't tell you which files it is looking at and which patterns it is trying from those files.

@larsks for visibility.

UASOP, 2019-05-02:
Thank you Peter!

I already have MavProxy installed. I will try the `set moddebug 3, module unload joystick` and keep troubleshooting. I will post any further progress and findings here.

UASOP, 2019-05-05:
Hey @peterbarker I have been continuing my attempts to add a new joystick with the instructions you mentioned in the previous comment. I have added the new YAML file for the interlink X, but when I "set moddebug 3" and "module load joystick". The output shows each YAML file that is checked. It is strange becuase the YAML file I added didn't show on the output. I tried reloading mavproxy with "sudo pip install Mavproxy"  but I am not sure that is the correct approach. The new YAML file is still not showing on the output. I have attached screen grabs and files below. I chanfed the YAML files to text in order to allow the file type to be uploaded.

[Terminal output.txt](https://github.com/ArduPilot/MAVProxy/files/3145074/Terminal.output.txt)

[GREAT PLANES InterLink-X.txt](https://github.com/ArduPilot/MAVProxy/files/3145071/GREAT.PLANES.InterLink-X.txt)
[great-planes.txt](https://github.com/ArduPilot/MAVProxy/files/3145072/great-planes.txt)

[image]

larsks, 2019-05-05:
@peterbarker thanks for the ping.

@UASOP  If you look at the terminal output you posted, your mavproxy is looking for joystick definition in `/usr/local/lib/python3.5/dist-packages/MAVProxy/modules/mavproxy_joystick/joysticks/`.  From the image you posted, it looks as if you've installed the file in `~/MAVProxy/MAVProxy/modules/mavproxy_joysticks/joysticks`.  Since you haven't installed the file where mavproxy is looking for it, it's not going to find it.

Fortunately, you don';t have to install it in either location: the joystick code will look in `$HOME/.mavproxy/joysticks` for additional joystick definitions. So put your file there, and try again, and let me know how it works out.

UASOP, 2019-05-06:
Thank you @larsks. I will give that a shot!

UASOP, 2019-05-06:
@peterbarker @larsks Thank you both so much for all of the help! II was looking for the file in the wrong location. You pointed me in the right direction @larsks. I have tested it out with some edits to the xbox 360 controller yaml and am currently testing out the new Great Planes Interlink-X YAML configuration, but so far it seems to be responding well. Can I post it here for any potential future use or for others if they should need it? I can post it the new Interlink-X file when I have tested it and ensured stability of results.  Thanks again.

larsks, 2019-05-06:
> Can I post it here for any potential future use or for others if they should need it?

Absolutely! But rather than just pasting it into this issue,  you should add it to the repository by submitting a  pull request. If you're not familiar with `git`, you can that via your browser by following [these instructions](https://help.github.com/en/articles/creating-new-files).

If that proves to difficult, you can paste it here and I will do approximately the same thing.

UASOP, 2019-06-10:
Thanks @larsks I think I have the controller setup and ready to go! Just trying to get the axis' mapped properly for hats. The three mode switches are showing up as axis. Once I get those set I will be submitting a pull request. I have one more question. How do I get the joystick module to load automatically. I have tried following the startup script documentation in mavproxy, but to no avail. I have also tried the --cmd="module load joystick;" and --cmd="module load joystick" in many different variations as well but it's not working. Are there any work arounds?

UASOP, 2019-11-11:
Here is the finished interlink-X yaml file. Tested and working. 
[great-planes Interlink-X.yaml.txt](https://github.com/ArduPilot/MAVProxy/files/3832569/great-planes.Interlink-X.yaml.txt)

stephendade, 2019-11-13:
@UASOP - can you put this in a Pull Request?

JamesM9, 2020-05-18:
@stephendade I have submitted a pull request as requested. :)  (https://github.com/JamesM9/MAVProxy/pull/1)

stephendade, 2020-05-18:
@JamesM9: You'll need to open that PR against the ArduPilot MAVProxy repository. You've currently got it open against your own repository.

JamesM9, 2020-05-19:
@stephendade whoops! my apologies. This is my first time doing this. Thanks for the correction. I have now submitted the pull request in the proper place. Thanks for the help!  https://github.com/ArduPilot/MAVProxy/pull/767

closed 2020-05-19 by UASOP