ArduPilot/MAVProxy #624: Performance considerations

746865646f6374, 2019-05-06:
Are there any performance considerations when I comes to the number of concurrent connections? If I use mavproxy to connect to the autopilot and open up 20 ports and have a bunch of connections established will that have a significant performance impact on the system?

wx4cb, 2019-05-06:
I would think that would mostly depend on the machine running mavproxy and
the network it's connected two

peterbarker, 2019-05-10:
On Mon, 6 May 2019, 746865646f6374 wrote:

> Are there any performance considerations when I comes to the number of concurrent connections? If I use mavproxy
> to connect to the autopilot and open up 20 ports and have a bunch of connections established will that have a
> significant performance impact on the system?

Yep, you may find MAVProxy struggles with this sort of thing.

If you are doing bog-standard dissemination of packets you might like to 
try mavlink-router or cmavnode (there are other similar projects out 
there).