ArduPilot/MAVProxy #630: map: "set home(altitude)" breake RTL

AndKe, 2019-05-30:
To reproduce: 
1: take off with AC, in map, right-click and set home(NO altitude), or try RTL after moving the copter.
2: observe that RTL works as expected.
3:  right click in map, select set home WITH altitude.
4: observe that RTL will just land in spot, - even is set with no altitude later on, the AC will only land , not RTL anymore.
- this is likely to be an issue with MAVProxy *and* ArduCopter

AndKe, 2019-05-30:
(safety risk)

tridge, 2019-05-30:
i wasn't able to reproduce this. Can you send tlog and onboard log from same flight showing the issue?

AndKe, 2019-05-31:
I have attached all meterial at: https://drive.google.com/open?id=1dUBeRX2a2qEn8Hm-khyZHazhqd2FBn-j

please watch the video, please note that RTL works perfectly until 1:48 where I use the "Set Home With height" feature, then RTL will not navigate anywhere anymore, even after new MAV_CMD_DO_SET_HOME