ArduPilot/MAVProxy #632: [Feature request] Add GPSD support

sgofferj, 2019-06-02:
Knowing its own location is useful for MAVproxy on a companion computer as well as when used as GCS. Connecting to a middleware layer  for reading GPS data has various advantages such as hardware-independence as well as the possibility of sharing the GPS device among various applications.
To my knowledge, the defacto standard under Linux and most POSIX OSs is GPSD (https://gpsd.gitlab.io/gpsd/index.html). Windows versions or compatible derivatives seem to be available too.
Thus I'd like to suggest a module which talks to GPSD for determining the own position.

wx4cb, 2019-06-02:
yup, this is technically possible, and if there wSA such a module, i would suggest using the shared memory process instead of the tcp/udp link. That being said, it's also possible that it would already know it's own location as you can just record the arming location as home. unless your GCS is 100s of miles away :D

sgofferj, 2019-06-02:
The arming location is not necessarily the current GCS location. Think e.g. mobile GCS. Or a stationary GCS away from the arming position. It doesn't need to be 100s of miles. Just a couple of 10 meters can make quite a difference when using MAVProxy also as an antenna tracker. Of course, things like mobile GCS are not everyday use cases but to be honest, MAVProxy also doesn't seem to be the GCS of choice for everyday use cases ;).