ArduPilot/MAVProxy #637: Can't get joystick to work with MAVProxy

imfatant, 2019-06-12:
I cannot get my Taranis joystick (which works fine with Mission Planner) to work with MAVProxy:

SIM_VEHICLE: Run ArduPlane
SIM_VEHICLE: "/home/anthony/git/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduPlane" "/home/anthony/git/ardupilot/build/sitl/bin/arduplane" "-S" "-I0" "--home" "-35.363261,149.165230,584,353" "--model" "xplane" "--speedup" "1"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--map" "--console"
RiTW: Starting ArduPlane : /home/anthony/git/ardupilot/build/sitl/bin/arduplane -S -I0 --home -35.363261,149.165230,584,353 --model xplane --speedup 1
xterm: cannot load font "10x20"
xterm: cannot load font "-Misc-Fixed-bold-R-*-*-13-120-75-75-C-60-ISO8859-1"
xterm: cannot load font "-Misc-Fixed-medium-R-*-*-13-120-75-75-C-120-ISO10646-1"
xterm: cannot load font "-Misc-Fixed-bold-R-*-*-13-120-75-75-C-120-ISO10646-1"
xterm: cannot load font "-misc-fixed-medium-r-normal--20-200-75-75-c-100-iso10646-1"
Connect tcp:127.0.0.1:5760 source_system=255
Loaded module console
Loaded module map
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> module load joystick
MAV> pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
MAVProxy.modules.mavproxy_joystick: Failed to find matching joystick.

Then..

MAV> set moddebug 3
MAV> module unload joystick
MAV> Unloaded module joystick
module load joystick
MAV> MAVProxy.modules.mavproxy_joystick: Initializing pygame
MAVProxy.modules.mavproxy_joystick: Initializing commands
MAVProxy.modules.mavproxy_joystick: Loading joystick definitions
MAVProxy.modules.mavproxy_joystick: Looking for joystick definitions in /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/Logitech_WingMan_3D.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/carolbox-usb.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/cypress-spektrum.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/great-planes.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/logicool-f310.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/logitech-dual-action.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/saili-simulator.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/sony-playstation.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/taranis.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/xbox-360.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from /home/anthony/.local/lib/python2.7/site-packages/MAVProxy/modules/mavproxy_joystick/joysticks/xboxdrv-mode2.yml
MAVProxy.modules.mavproxy_joystick: Failed to find matching joystick.
Loaded module joystick

Don't know what to do next. Please help. Been stuck on this for three days. :(

stephendade, 2019-06-12:
There's a joystick information script in MAVProxy here: https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_joystick/findjoy.py

It'll tell you if your joystick is being properly detected or not.

imfatant, 2019-06-12:
Here's what happens:

anthony@DESKTOP-DJ821LK:~$ python -m MAVProxy.modules.mavproxy_joystick.findjoy
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
This script will help you identify axis, button, hat, and
ball ids on your joystick in order to create new joystick
definitions for the mavproxy_joystick module.

Available joysticks:

Select a joystick (L to list, Q to quit)

As you can see, nothing is listed. Not sure where to go next.

stephendade, 2019-06-13:
hmm .... that's a strange one.

For some reason your joystick is not being detected. Can you try with other joysticks? On different computers?

imfatant, 2019-06-13:
Tried 'native Windows' MAVProxy:

MANUAL> module load joystick
MANUAL> pygame 1.9.5
Hello from the pygame community. https://www.pygame.org/contribute.html
MAVProxy.modules.mavproxy_joystick: Initializing pygame
MAVProxy.modules.mavproxy_joystick: Initializing commands
MAVProxy.modules.mavproxy_joystick: Loading joystick definitions
MAVProxy.modules.mavproxy_joystick: Looking for joystick definitions in C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks
MAVProxy.modules.mavproxy_joystick: Found joystick (FrSky Taranis Joystick)
MAVProxy.modules.mavproxy_joystick: Failed to find matching joystick.
Loaded module joystick

Still no joy.

stephendade, 2019-06-14:
>  MAVProxy.modules.mavproxy_joystick: Found joystick (FrSky Taranis Joystick)

Well, it found the joystick there so that's something.

It could be that the latest FrSky drivers have renamed the joystick so MAVProxy is having trouble mtching it. Currently it tries to look for ``FrSky FrSky Taranis*``, which is a slightly different name to your joystick.

Try editing the name in ``./MAVProxy/modules/mavproxy_joystick/taranis.yml`` to your detected name: ``FrSky Taranis`` and then see if it works.

imfatant, 2019-06-16:
MAVProxy.modules.mavproxy_joystick: Looking for joystick definitions in C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick

I used the Windows 10 executable installer. The directory doesn't exist. I tried a reinstall. Still doesn't exist. Hmm.

imfatant, 2019-06-16:
Changed `FrSky FrSky Taranis*` to `FrSky Taranis` and got the same problem.

Then changed it to `FrSky Taranis Joystick` and this happens:

MANUAL> module load joystick
MANUAL> pygame 1.9.5
Hello from the pygame community. https://www.pygame.org/contribute.html
MAVProxy.modules.mavproxy_joystick: Initializing pygame
MAVProxy.modules.mavproxy_joystick: Initializing commands
MAVProxy.modules.mavproxy_joystick: Loading joystick definitions
MAVProxy.modules.mavproxy_joystick: Looking for joystick definitions in C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\carolbox-usb.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\cypress-spektrum.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\great-planes.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\logicool-f310.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\logitech-dual-action.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\Logitech_WingMan_3D.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\saili-simulator.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\sony-playstation.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\taranis.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\xbox-360.yml
MAVProxy.modules.mavproxy_joystick: Loading definition from C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\xboxdrv-mode2.yml
MAVProxy.modules.mavproxy_joystick: Found joystick (FrSky Taranis Joystick)
MAVProxy.modules.mavproxy_joystick: Using C:\Program Files (x86)\MAVProxy\MAVProxy\modules\mavproxy_joystick\joysticks\taranis.yml ("FrSky Taranis Joystick" matches pattern "FrSky Taranis Joystick")
Loaded module joystick
Traceback (most recent call last):
  File "MAVProxy\mavproxy.py", line 809, in periodic_tasks
  File "MAVProxy\modules\mavproxy_joystick\__init__.py", line 146, in idle_task
pygame.error: Invalid joystick axis

imfatant, 2019-06-16:
Solved the previous issue by truncating taranis.yml to:

description: >
 FrSky Taranis transmitter as a joystick via USB cable

match:
  - 'FrSky Taranis Joystick'

controls:
  - channel: 1
    type: axis
    id: 0
  - channel: 2
    type: axis
    id: 1
  - channel: 3
    type: axis
    id: 2
  - channel: 4
    type: axis
    id: 3
  - channel: 5
    type: axis
    id: 4
  - channel: 6
    type: axis
    id: 5

Throttle, aileron and elevator are working when ArduPlane is switched into MANUAL, but not smoothly. Rudder is not working.