ArduPilot/MAVProxy #645: Download and Installation - Raspberry Pi

jchidley, 2019-07-11:
URL: ../getting_started/download_and_installation
Title: Download and Installation

I needed to install python3-lxml (possibly also python-lxml) to get this to work on a Raspberry Pi

jchidley, 2019-07-25:
Actually, I needed to change 
this:
sudo apt-get install python3-dev python3-opencv python3-wxgtk3.0 python3-pip python3-matplotlib python3-pygame python3-lxml python3-yaml
to this:
sudo apt-get install python-dev python-opencv python-wxgtk3.0 python-pip python-matplotlib python-pygame python-lxml python-yaml

mhallak, 2020-05-20:
Is there an update on this issue?

stephendade, 2020-09-06:
The ArduPilot wiki now has the correct install instructions: https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html

closed 2020-09-06 by stephendade