ArduPilot/MAVProxy #685: MacOS: notarizing required for Catalina/Jan 2020

auturgy, 2019-09-06:
https://developer.apple.com/news/?id=09032019a
This is going to cause problems for install to MacOS come Jan 2020.

peterbarker, 2019-09-07:
On Fri, 6 Sep 2019, auturgy wrote:

> https://developer.apple.com/news/?id=09032019a
> This is going to cause problems for install to MacOS come Jan 2020.

Assuming we weren't using native code - does it affect MAVProxy?

auturgy, 2019-09-07:
Unknown, just making people aware. The Apple policy seems to target binaries, so I think interpreted code should be fine so long as the interpreter and any binary dependencies are notarised. Something to test I guess.

monkeypants, 2019-09-07:
Catalina doesn't come with a python interpreter. Maybe distributing MAVProxy to that platform will require bundling a python interpreter (and all the dependencies, libraries etc) into a package then getting the whole thing blessed with their latest ritual.

We have an Apple build server at work doing a similar sort of thing for some iOS apps (build > package > bless > distribute).

srmainwaring, 2020-11-08:
I can confirm a standard brew python installation is sufficient to run MAVProxy on macOS Catalina if you install MAVProxy from the master branch.

If you are using Python 3.9 (which is now the default in brew) you may need to install pymavlink from source as well while recent changes propagate through to the released packages (there is a patch to `pymavlink/generator/mavcrc.py` to account for array.fromstring being removed in Python 3.9).

srmainwaring, 2023-04-25:
Closing - the notarising requirement does not appear to have affected MAVProxy in versions since Catalina.

closed 2023-04-25 by srmainwaring