ArduPilot/MAVProxy #692: Failed to load module: No module named 'console'.

machenxiang, 2019-09-20:
Ubuntu 16.04
python Python 2.7.16 |Anaconda,
when i ../Tools/autotest/sim_vehicle.py -f gazebo-iris --console --map,it occurs this error.

SIM_VEHICLE: Start
SIM_VEHICLE: Killing tasks
SIM_VEHICLE: Starting up at SITL location
SIM_VEHICLE: WAF build
SIM_VEHICLE: Configure waf
SIM_VEHICLE: "/home/mcx/ardupilot/modules/waf/waf-light" "configure" "--board" "sitl"
Setting top to                           : /home/mcx/ardupilot 
Setting out to                           : /home/mcx/ardupilot/build 
Autoconfiguration                        : enabled 
Setting board to                         : sitl 
Using toolchain                          : native 
Checking for 'g++' (C++ compiler)        : /usr/lib/ccache/g++ 
Checking for 'gcc' (C compiler)          : /usr/lib/ccache/gcc 
Checking for c flags '-MMD'              : yes 
Checking for cxx flags '-MMD'            : yes 
Checking for need to link with librt     : not necessary 
Checking for feenableexcept              : yes 
Checking for HAVE_CMATH_ISFINITE         : yes 
Checking for HAVE_CMATH_ISINF            : yes 
Checking for HAVE_CMATH_ISNAN            : yes 
Checking for NEED_CMATH_ISFINITE_STD_NAMESPACE : yes 
Checking for NEED_CMATH_ISINF_STD_NAMESPACE    : yes 
Checking for NEED_CMATH_ISNAN_STD_NAMESPACE    : yes 
Checking for header endian.h                   : yes 
Checking for header byteswap.h                 : yes 
Checking for HAVE_MEMRCHR                      : yes 
Checking for program 'python'                  : /home/mcx/anaconda3/envs/py2.7/bin/python 
Checking for python version >= 2.7.0           : 2.7.16 
Checking for program 'python'                  : /home/mcx/anaconda3/envs/py2.7/bin/python 
Checking for python version >= 2.7.0           : 2.7.16 
Source is git repository                       : yes 
Update submodules                              : yes 
Checking for program 'git'                     : /usr/bin/git 
Checking for program 'size'                    : /usr/bin/size 
Benchmarks                                     : disabled 
Unit tests                                     : enabled 
Scripting                                      : enabled 
Scripting runtime checks                       : enabled 
Checking for program 'rsync'                   : /usr/bin/rsync 
'configure' finished successfully (1.197s)
SIM_VEHICLE: Building
SIM_VEHICLE: "/home/mcx/ardupilot/modules/waf/waf-light" "build" "--target" "bin/arducopter"
Waf: Entering directory `/home/mcx/ardupilot/build/sitl'
Embedding file sandbox.lua:libraries/AP_Scripting/scripts/sandbox.lua
Waf: Leaving directory `/home/mcx/ardupilot/build/sitl'

BUILD SUMMARY
Build directory: /home/mcx/ardupilot/build/sitl
Target          Text     Data  BSS    Total  
---------------------------------------------
bin/arducopter  2290607  2920  65216  2358743

Build commands will be stored in build/sitl/compile_commands.json
'build' finished successfully (1.960s)
SIM_VEHICLE: Using defaults from (/home/mcx/ardupilot/Tools/autotest/default_params/copter.parm,/home/mcx/ardupilot/Tools/autotest/default_params/gazebo-iris.parm)
SIM_VEHICLE: Run ArduCopter
SIM_VEHICLE: "/home/mcx/ardupilot/Tools/autotest/run_in_terminal_window.sh" "ArduCopter" "/home/mcx/ardupilot/build/sitl/bin/arducopter" "-S" "-I0" "--model" "gazebo-iris" "--speedup" "1" "--defaults" "/home/mcx/ardupilot/Tools/autotest/default_params/copter.parm,/home/mcx/ardupilot/Tools/autotest/default_params/gazebo-iris.parm"
SIM_VEHICLE: Run MavProxy
SIM_VEHICLE: "mavproxy.py" "--master" "tcp:127.0.0.1:5760" "--sitl" "127.0.0.1:5501" "--out" "127.0.0.1:14550" "--out" "127.0.0.1:14551" "--map" "--console"
RiTW: Starting ArduCopter : /home/mcx/ardupilot/build/sitl/bin/arducopter -S -I0 --model gazebo-iris --speedup 1 --defaults /home/mcx/ardupilot/Tools/autotest/default_params/copter.parm,/home/mcx/ardupilot/Tools/autotest/default_params/gazebo-iris.parm
/home/mcx/anaconda3/bin/mavproxy.py:20: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  from imp import reload
Connect tcp:127.0.0.1:5760 source_system=255
Failed to load module: No module named 'console'. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Failed to load module: No module named 'map'. Use 'set moddebug 3' in the MAVProxy console to enable traceback
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from tcp:127.0.0.1:5760
MAV> Process Process-1:
Traceback (most recent call last):
  File "/home/mcx/anaconda3/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
  File "/home/mcx/anaconda3/lib/python3.7/multiprocessing/process.py", line 99, in run
    self._target(*self._args, **self._kwargs)
  File "/home/mcx/anaconda3/lib/python3.7/site-packages/MAVProxy/modules/mavproxy_map/srtm.py", line 173, in createFileListHTTP
    parser.feed(data)
  File "/home/mcx/anaconda3/lib/python3.7/html/parser.py", line 110, in feed
    self.rawdata = self.rawdata + data
TypeError: can only concatenate str (not "bytes") to str

stephendade, 2019-09-20:
>  Checking for program 'python' : /home/mcx/anaconda3/envs/py2.7/bin/python

Ok, Python 2.7 here in the ardupilot build process

> File "/home/mcx/anaconda3/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
self.run()

You're running Python 3.7 here when executing MAVProxy. Is your environment properly configured?

>  File "/home/mcx/anaconda3/lib/python3.7/html/parser.py", line 110, in feed
>  self.rawdata = self.rawdata + data
>  TypeError: can only concatenate str (not "bytes") to str

Try using the latest MAVProxy from Github, I think this was fixed in 970c4fdc0b1c60f144380ef7f7e40cdad53d67dc

machenxiang, 2019-09-20:
> Ok, Python 2.7 here in the ardupilot build process
>
>
> You're running Python 3.7 here when executing MAVProxy. Is your environment properly configured?
>
>
> Try using the latest MAVProxy from Github, I [...]

Thanks for your reply . I have virtual python environment,I use source activate  py2.7 to change the version of python.it know is python 2.7. But I will try to install latest Mavproxy .   Thank you again.

Anmol-2003, 2023-05-19:
im facing the same issue where it says - No module named 'map' and 'console' and 'set moddebug 3' for traceback. 
I'm using a conda env with python 3.7.16 with all the essential libraries and modules installed and i still dont know what is causing this issue 
Am using Ubuntu 20.0.4 LTS

Z-Xiong, 2023-06-02:
I solved it.
use "./sim_vehicle.py -v ArduCopter --console --map --moddebug 3" instead of "set moddebug 3"
You will see what module is missing.
Just pip install XX.