ArduPilot/MAVProxy #709: MAVProxy sending invalid terrain data above 60N

rmackay9, 2019-12-02:
I think that MAVProxy is sending invalid terrain data to the vehicle for latitudes above 60N for which the ardupilot SRTM database has no data.  This means that a user in finland (and many other places probably) could attempt to fly a terrain altitude mission and all would appear fine until the vehicle crashed into the ground.

This can be reproduced in SITL by doing the following:

- edit the Tools/autotest/Locations.txt file and add finland as "62.23075, 25.6993,0,0"
- ../Tools/autotest/sim_vehicle.py -L finland --map --console
- create a short mission with a takeoff, waypoint (with an altitude of 10m AGL), RTL
- fly the mission

The vehicle will fly the mission just as if it's a relative altitude.  I.e. it will not climb over mountains.

Here is the issue on the vehicle side: https://github.com/ArduPilot/ardupilot/issues/12923

rmackay9, 2019-12-03:
Just as a side note, Mission Planner (beta) has been fixed so that it no longer provides invalid altitudes if there is no SRTM data.

Ryanf55, 2024-05-08:
Is this still an issue?

rmackay9, 2024-05-08:
@Ryanf55,

It should be fixed but we should test..