ArduPilot/MAVProxy #716: ntrip module not working

petergerten, 2019-12-10:
In a clean Ubuntu 18.04 docker container with mavproxy installed via pip, I get the following error when trying to load the ntrip module:
```

MAV> module load ntrip
MAV> Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 559, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python3.6/dist-packages/MAVProxy/modules/mavproxy_ntrip.py", line 8, in <module>
    from MAVProxy.modules.lib import ntrip
  File "/usr/local/lib/python3.6/dist-packages/MAVProxy/modules/lib/ntrip.py", line 13, in <module>
    import rtcm3
ModuleNotFoundError: No module named 'rtcm3'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 405, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 562, in import_package
    mod = __import__(name)
  File "/usr/local/lib/python3.6/dist-packages/MAVProxy/modules/mavproxy_ntrip.py", line 8, in <module>
    from MAVProxy.modules.lib import ntrip
  File "/usr/local/lib/python3.6/dist-packages/MAVProxy/modules/lib/ntrip.py", line 13, in <module>
    import rtcm3
ModuleNotFoundError: No module named 'rtcm3'

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 559, in import_package
    mod = __import__(name)
ModuleNotFoundError: No module named 'ntrip'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/mavproxy.py", line 405, in load_module
    m = import_package(modpath)
  File "/usr/local/bin/mavproxy.py", line 562, in import_package
    mod = __import__(name)
ModuleNotFoundError: No module named 'ntrip'

Failed to load module: No module named 'ntrip'.
```
The rtcm3.py file actually does exist:
```
root@13314e1688b7:/# find / -name rtcm3.py
/usr/local/lib/python3.6/dist-packages/MAVProxy/modules/lib/rtcm3.py

```
Does anyone know what is going wrong here?

petergerten, 2019-12-10:
I could reproduce this when using git master instead of the pip repo as well.

rmackay9, 2019-12-10:
@petergerten, it's very interesting that you're trying to get an NTRIP server going though.  I've been thinking it would be great if we had instructions on the ArduPilot wiki about how to set one up.  I think with the Ublox F9 coming out and massively reducing the cost of L2 RTK there will be a significant number of people who suddenly want to setup an Ntrip server.

I guess my point is that if you figure this out I hope we can document it.

petergerten, 2019-12-10:
@rmackay9 I got the ntrip module working now, I just removed the "import rtcm3" line from ntrip.py.

Will follow up on how it goes :)

cjmssmd, 2020-01-14:
@petergerten any luck? Are you using your own ntrip caster or a public one?

cjmssmd, 2020-03-01:
@petergerten FYI after @tridge 's recent fix the ntrip module has started working for me. Not perfectly, yet (see my recent issue).  But connecting to the caster, receiving RTCM messages, sending them on to the autopilot.