ArduPilot/MAVProxy #727: AttributeError: 'MAVLink_statustext_message' object has no attribute 'chunk_seq'

sharkvary, 2020-02-01:
The following error occurs:

[env]
Ardupilot SITL version : arducopter v3.6.2
Mavproxy branch : master

Connect /dev/pts/3 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/pts/3
MAV> online system 1
STABILIZE> Mode STABILIZE
fence breach
Exception in thread main_loop:
Traceback (most recent call last):
File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
self.run()
File "/usr/lib/python3.6/threading.py", line 864, in run
self._target(*self._args, **self._kwargs)
File "/home/sharkvary/.local/lib/python3.6/site-packages/MAVProxy-1.8.18-py3.6.egg/EGG-INFO/scripts/mavproxy.py", line 996, in main_loop
File "/home/sharkvary/.local/lib/python3.6/site-packages/MAVProxy-1.8.18-py3.6.egg/EGG-INFO/scripts/mavproxy.py", line 698, in process_master
File "/home/sharkvary/.local/lib/python3.6/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 13475, in parse_buffer
m = self.parse_char("")
File "/home/sharkvary/.local/lib/python3.6/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 13411, in parse_char
self.__callbacks(m)
File "/home/sharkvary/.local/lib/python3.6/site-packages/pymavlink/dialects/v20/ardupilotmega.py", line 13388, in __callbacks
self.callback(msg, *self.callback_args, **self.callback_kwargs)
File "/home/sharkvary/.local/lib/python3.6/site-packages/MAVProxy-1.8.18-py3.6.egg/MAVProxy/modules/mavproxy_link.py", line 662, in master_callback
self.master_msg_handling(m, master)
File "/home/sharkvary/.local/lib/python3.6/site-packages/MAVProxy-1.8.18-py3.6.egg/MAVProxy/modules/mavproxy_link.py", line 513, in master_msg_handling
pending.add_chunk(m)
File "/home/sharkvary/.local/lib/python3.6/site-packages/MAVProxy-1.8.18-py3.6.egg/MAVProxy/modules/mavproxy_link.py", line 485, in add_chunk
self.chunks[m.chunk_seq] = m.text
AttributeError: 'MAVLink_statustext_message' object has no attribute 'chunk_seq'

peterbarker, 2020-02-01:
On Fri, 31 Jan 2020, JongWoon Lee wrote:

> The following error occurs:

You will need to upgrade your pymavlink install.

Peter

sharkvary, 2020-02-01:
> On Fri, 31 Jan 2020, JongWoon Lee wrote: The following error occurs:
> You will need to upgrade your pymavlink install. Peter

Here is my pymavlink version information. Despite the latest version still not.

Name: pymavlink
Version: 2.4.4
Summary: Python MAVLink code
Home-page: https://github.com/ArduPilot/pymavlink/
Author: UNKNOWN
Author-email: UNKNOWN
License: LGPLv3
Location: /home/sharkvary/.local/lib/python3.6/site-packages
Requires: future, lxml

peterbarker, 2020-02-01:
On Fri, 31 Jan 2020, JongWoon Lee wrote:

> Here is my pymavlink version information. Despite the latest version still not.

What version of the message definitions are you using when installing?

sharkvary, 2020-02-01:
> On Fri, 31 Jan 2020, JongWoon Lee wrote: Here is my pymavlink version information. Despite the latest version still not.
> What version of the message definitions are you using when installing?

how to check it? I was just starting Python.

sharkvary, 2020-02-03:
> On Fri, 31 Jan 2020, JongWoon Lee wrote: Here is my pymavlink version information. Despite the latest version still not.
> What version of the message definitions are you using when installing?

What version do I need? And how to check it? I was just starting Python.

marselomeri, 2020-02-03:
Hi, 
I had the same problem. I fixed it by updating pymavlink, the updated version is 2.4.6 , not 2.4.4. 
Just use this command and will automatically check for the update sudo pip2 install -U pymavlink.

sharkvary, 2020-02-04:
> Hi,
> I had the same problem. I fixed it by updating pymavlink, the updated version is 2.4.6 , not 2.4.4.
> Just use this command and will automatically check for the update sudo pip2 install -U pymavlink.

Thank you~! I fixed it with what you told me.

closed 2020-02-04 by sharkvary