ArduPilot/MAVProxy #735: ntrip generating RTCM messages @ about 1-2 Hz, less than mission planner @ about 7-12 Hz

cjmssmd, 2020-03-01:
The ntrip module is not passing on as many RTCM messages as mission planner. 

connecting the ntrip module (Vehicle 255 Comp 0 in the MAVlink inspector screen shot below) and mission planner (Vehicle 255 Comp 190) to the same caster - cddis-caster.gsfc.nasa.gov:443 caster (contact me for credentials), I see:

[image]

The effect of this is that my GPS doesn't maintain RTK fix when I'm using solely ntrip.

cjmssmd, 2020-03-01:
In the screenprint above, see Vehicle 255 Comp 0 (which is the mavproxy GCS running the ntrip module) is generating GPS_RTCM_DATA messages at 0.7 Hz, 54 bps, while Comp 190 (which is mission planner) is generating 6.3 Hz and 1203 bps.   Both are connected to the same caster.

highfreq, 2020-03-02:
Same problem here. Ntrip module works for few minutes than it goes to 3d fix and never goes back to float or fixed. With mission planner or another ntrip proxy it works and stays up.
On my system i also see very low data rate from ntrip module, below 1hz.

highfreq, 2020-03-02:
I wonder if heartbeat set to zero in my mavinit script has anything to do with it.

cjmssmd, 2020-03-02:
I doubt it. My heartbeat is not zero and I see the same behavior. I suspect a bug in the ntrip code that is causing some messages to not be sent. Perhaps longer messages that need to be broken up into parts (as the GPS_RTCM_DATA message has a fixed length), or, something in the checksum logic for longer messages. In any case I suspect it may be faster for someone more familiar with the code to debug now that we have identified the problem.

cjmssmd, 2020-03-03:
I ran the ntrip.py program standalone (i.e. from the command line and not as part of a module within mavproxy) and it receives 3.5-5.5 RTCM messages/sec. But the same program when run as a module from within mavproxy only delivers 0.9 messages/sec. There may be multiple issues going on.

cjmssmd, 2020-03-03:
This issue turned out to not be caused by the ntrip module but another module (my custom console module) that was not behaving well, i.e. too much processing inside the idle_task method of the other module.  When I fixed that other module, my ntrip module started sending GPS_RTCM_DATA messages just fine.  So, I'm closing this issue with a "never mind".

closed 2020-03-03 by cjmssmd

peterbarker, 2020-03-10:
On Mon, 2 Mar 2020, Chris wrote:

> This issue turned out to not be caused by the ntrip module but another module (my custom console module) that was not behaving well, i.e. too much
> processing inside the idle_task method of the other module. When I fixed that other module, my ntrip module started sending GPS_RTCM_DATA messages just

That should be detectable, and we should warn people if it ever happens. 
Would have saved you time here.

cjmssmd, 2020-03-10:
What would have saved me time is a prompt in the documentation to look at the GPS_RTCM_DATA message volume as a way of diagnosing RTCM.    Once I noticed this it was easy to fix.  But it took me forever to figure out why the gps receiver was acquiring and losing the RTCM lock so often.   

Now that it's fixed the gps receiver holds a lock much more of the time. But not 100% of the time.

Incidentally the way I'm handling an interruption in RTCM data (or other RTCM problems that cause the GPS lock to fail, such as for example the ntrip server itself being down) is further downstream.  I'm detecting when GPS lock is lost (i.e. GPS fix changes from >=5 to <5) and, in my case, which is a rover, going into HOLD mode until it recovers, and then going back into AUTO mode.