ArduPilot/MAVProxy #736: Install MAVProxy on mac with Pipenv fail

emialonzo, 2020-03-04:
Hi, I try to install the **MAVProxy** with pipenv and alway fail when locking packages dependencies. 
Below I paste the command output.

**$** pipenv install wxPython gnureadline billiard numpy pyparsing MAVProxy
Warning: the environment variable LANG is not set!
We recommend setting this in ~/.profile (or equivalent) for proper expected behavior.
Creating a virtualenv for this project…
Pipfile: /Users/dronfies/desarrollo/mavlinkProxy/Pipfile
Using /usr/local/Cellar/pipenv/2018.11.26_3/libexec/bin/python3.8 (3.8.1) to create virtualenv…
⠴ Creating virtual environment...Already using interpreter /usr/local/Cellar/pipenv/2018.11.26_3/libexec/bin/python3.8
Using real prefix '/usr/local/Cellar/python@3.8/3.8.1/Frameworks/Python.framework/Versions/3.8'
New python executable in /Users/dronfies/.local/share/virtualenvs/mavlinkProxy-TTTNJ-vn/bin/python3.8
Also creating executable in /Users/dronfies/.local/share/virtualenvs/mavlinkProxy-TTTNJ-vn/bin/python
Installing setuptools, pip, wheel...
done.

✔ Successfully created virtual environment! 
Virtualenv location: /Users/dronfies/.local/share/virtualenvs/mavlinkProxy-TTTNJ-vn
Creating a Pipfile for this project…
Installing wxPython…
Adding wxPython to Pipfile's [packages]…
✔ Installation Succeeded 
Installing gnureadline…
Adding gnureadline to Pipfile's [packages]…
✔ Installation Succeeded 
Installing billiard…
Adding billiard to Pipfile's [packages]…
✔ Installation Succeeded 
Installing numpy…
Adding numpy to Pipfile's [packages]…
✔ Installation Succeeded 
Installing pyparsing…
Adding pyparsing to Pipfile's [packages]…
✔ Installation Succeeded 
Installing MAVProxy…
Adding MAVProxy to Pipfile's [packages]…
✔ Installation Succeeded 
Pipfile.lock not found, creating…
Locking [dev-packages] dependencies…
Locking [packages] dependencies…

Traceback (most recent call last):
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/pexpect/expect.py", line 109, in expect_loop
    return self.timeout()
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect.popen_spawn.PopenSpawn object at 0x111e8d040>
searcher: searcher_re:
    0: re.compile('\n')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/bin/pipenv", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/decorators.py", line 64, in new_func
    return ctx.invoke(f, obj, *args, **kwargs)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/cli/command.py", line 235, in install
    retcode = do_install(
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/core.py", line 1983, in do_install
    do_init(
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/core.py", line 1239, in do_init
    do_lock(
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/core.py", line 1058, in do_lock
    venv_resolve_deps(
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/utils.py", line 649, in venv_resolve_deps
    c = resolve(cmd, sp)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/utils.py", line 517, in resolve
    result = c.expect(u"\n", timeout=environments.PIPENV_INSTALL_TIMEOUT)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/delegator.py", line 215, in expect
    self.subprocess.expect(pattern=pattern, timeout=timeout)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/pexpect/spawnbase.py", line 340, in expect
    return self.expect_list(compiled_pattern_list,
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/pexpect/spawnbase.py", line 369, in expect_list
    return exp.expect_loop(timeout)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/pexpect/expect.py", line 119, in expect_loop
    return self.timeout(e)
  File "/usr/local/Cellar/pipenv/2018.11.26_3/libexec/lib/python3.8/site-packages/pipenv/vendor/pexpect/expect.py", line 82, in timeout
    raise TIMEOUT(msg)
pexpect.exceptions.TIMEOUT: <pexpect.popen_spawn.PopenSpawn object at 0x111e8d040>
searcher: searcher_re:
    0: re.compile('\n')
<pexpect.popen_spawn.PopenSpawn object at 0x111e8d040>
searcher: searcher_re:
    0: re.compile('\n')

srmainwaring, 2020-12-06:
@emialonzo I am not familiar with `pipenv` but succeeded in installing it and the latest version of MAVProxy. Here are my steps and console outputs:

**System**

- macOS Catalina 10.15.7
- brew (fully up to date)
- Python 3.9.0 (`/usr/local/opt/python3/libexec/bin` is prefixed to `PATH` in my `~/.zprofile`) 

**Install pipenv**

```bash
brew install pipenv
==> Downloading https://homebrew.bintray.com/bottles/pipenv-2020.11.4.catalina.b
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/e1f59430bce27f5a9fbc4
######################################################################## 100.0%
==> Pouring pipenv-2020.11.4.catalina.bottle.tar.gz
==> Caveats
Bash completion has been installed to:
  /usr/local/etc/bash_completion.d

zsh completions have been installed to:
  /usr/local/share/zsh/site-functions
==> Summary
🍺  /usr/local/Cellar/pipenv/2020.11.4: 1,862 files, 25.2MB
``` 

***Install MAVProxy***

```bash
#create workspace folder
$ mkdir issue_736 && cd issue_736

#install pre-requisites
$ pipenv install wxPython                               
Creating a Pipfile for this project...
Installing wxPython...
Adding wxPython to Pipfile's [packages]...
✔ Installation Succeeded 
Pipfile.lock not found, creating...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
✔ Success! 
Updated Pipfile.lock (eb1416)!
Installing dependencies from Pipfile.lock (eb1416)...
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.

$ pipenv install gnureadline
Installing gnureadline...
Adding gnureadline to Pipfile's [packages]...
✔ Installation Succeeded 
Pipfile.lock (eb1416) out of date, updating to (48a79c)...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
✔ Success! 
Updated Pipfile.lock (48a79c)!
Installing dependencies from Pipfile.lock (48a79c)...
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.

$ pipenv install numpy 
Installing numpy...
Adding numpy to Pipfile's [packages]...
✔ Installation Succeeded 
Pipfile.lock (48a79c) out of date, updating to (07e46e)...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
✔ Success! 
Updated Pipfile.lock (07e46e)!
Installing dependencies from Pipfile.lock (07e46e)...
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.

$ pipenv install pyparsing
Installing pyparsing...
Adding pyparsing to Pipfile's [packages]...
✔ Installation Succeeded 
Pipfile.lock (07e46e) out of date, updating to (d80a55)...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
✔ Success! 
Updated Pipfile.lock (d80a55)!
Installing dependencies from Pipfile.lock (d80a55)...
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.

pipenv install MAVProxy 
Installing MAVProxy...
Adding MAVProxy to Pipfile's [packages]...
✔ Installation Succeeded 
Pipfile.lock (d80a55) out of date, updating to (1945f0)...
Locking [dev-packages] dependencies...
Locking [packages] dependencies...
Building requirements...
Resolving dependencies...
✔ Success! 
Updated Pipfile.lock (1945f0)!
Installing dependencies from Pipfile.lock (1945f0)...
  🐍   ▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉▉ 0/0 — 00:00:00
To activate this project's virtualenv, run pipenv shell.
Alternatively, run a command inside the virtualenv with pipenv run.
```

I did see an error message when I tried to install a particular version of `numpy` (`numpy==1.19.3`), but for macOS the current version works fine.

***Run MAVProxy***

```bash
pipenv run MAVProxy.py
Auto-detected serial ports are:
Connect 0.0.0.0:14550 source_system=255
Running script (/Users/rhys/.mavinit.scr)
Running script /Users/rhys/.mavinit.scr
Log Directory: 
Telemetry log: mav.tlog
Waiting for heartbeat from 0.0.0.0:14550
MAV> link 1 down
```

I use the global python and pip instances to install to `/usr/local/lib/python3.9/site-packages/` rather than virtual envs as I also use ROS and have not had much success using `pyenv` or similar with my local ROS installation. The installation is then straightforward and follows the wiki entry: [Download and Installation: Mac](https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html#mac).

If you are using Python 3.8 or later, `billiard` is no longer used in MAVProxy and in fact will not work on macOS Catalina.

srmainwaring, 2023-04-25:
Closing due to no activity.

closed 2023-04-25 by srmainwaring