ArduPilot/MAVProxy #746: How can i connect MAVProxy (in server) to QgroundControl (in local)

Mazvangui, 2020-03-27:
I need to connect MAVProxy running in my server to QgroundControl in my local, so i tried to do this way:

1. Server side:
- run DroneKit: `dronekit-sitl copter` (by default listen to 5760 port)
- run MAVProxy: `mavproxy.py --quadcopter --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551`
- add my local output to MAVProxy: `output add my_local_ip_addr:14550`

2. Local side:

- Just running Qgroundcontrol (who listen by default to 14550 port).
**But no connexion shown from my QgroundControl.**
- I also tried to send command to my SITL from mavproxy:
`python simple_goto.py --connect udp:my_server_ip_addr:14550` 
** But i have a connexion issue **

stephendade, 2020-03-30:
Is MAVProxy showing that it's connected to the SITL instance?

THOM-AwS, 2024-08-31:
127.0.0.1 is the address of `this computer`. that means that you are telling mavproxy to advertise your application only on the server, and your local will never be able to get to it. if you want to advertise to things outside your server, you need to change 127.0.0.1 to 0.0.0.0 or whatever address you want to connect, like the ip of your local. 

if you are doing all this inside a lan, 0.0.0.0 is fine.

probably try to understand network security before you go leaving that on all the time.